Koenkk/zigbee2mqtt #27062: Bug: illuminance attribute returns null for ZY-M100 after upgrade to 2.2.1-1

bassani72, 2025-04-09:
### What happened?

**Affected Version**: Zigbee2MQTT 2.2.1-1  
**Working Version**: Zigbee2MQTT 2.2.0 -1
**Device**: Tuya ZY-M100 (TS0601 / _TZE204_qasjif9e)

After updating to version 2.2.1-1, all my ZY-M100 sensors stopped reporting the `illuminance` attribute via MQTT. Instead of a numeric value, I only receive Illuminance: "null" 

### What did you expect to happen?

The rest of the sensor data (e.g., target_distance, radar_sensitivity, presence, etc.) continues to work correctly.

I tested:

- Full reinterview of the devices ✅
- Clean `state.json` and `database.db` ✅
- Manual attribute reading ✅
- Confirmed cluster `manuSpecificTuya` is present ✅
- Restoring Zigbee2MQTT 2.2.0 → `illuminance` works again ✅

This is likely a regression in how `manuSpecificTuya` or exposes are handled in 2.2.1-1.

Please let me know if I can provide logs or further testing.

Thanks!

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

- 7.4.4 [GA]

### Adapter

EmberZNet 

### Setup

Add-on on Home Assistant OS, running on Virtual Machine Vmware

bassani72, 2025-04-10:
> Already fixed in dev
>
> Changes will be available in the dev branch<https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html> in a few hours from now and in the next release which is every 1st of the month.
>
>
>
> Already fixed in dev
>
> Changes will be available in the dev branch<https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html> in a few hours from now and in the next release which is every 1st of the month.

Perfect, thanks!

closed 2025-04-10 by Koenkk

bassani72, 2025-04-22:
HI, yesterday I updated tio 2.2.1-1 and the bug is still present.
I had to move back 2.2.0-1 to function again...
Let me know
Thanks
Luca

lukaszzyla, 2025-04-29:
For me illuminance works on both stable and edge HA addons.
check my problem here:
https://github.com/Koenkk/zigbee2mqtt/issues/27212
I did install 2.2.0 in docker as recommended but then presence is always true (frozen in true opposed to frozen in null)
Any idea if this is the same issue?
I only bought this sensor and was wondering if it works as motion sensor at all?

naevtamarkus, 2025-05-05:
Hi all! Just wanted to say that 2.3.0 fixed it for me. Thanks!

bassani72, 2025-05-05:
Also for me for TY-M100 illuminance, thanks!

lukaszzyla, 2025-05-06:
I am in the unlucky team then, because for me it is still the same.
The funny thing is that illuminance was always working for me together with distance.
but the presence is either true or none
I have removed and readded the device so many times an many different versions of z2m.
I was hoping that 2.3.0 fixes it for me like for rest of you but unfortunatelly - no difference.
mqtt broker:

[image]

z2m:
When I add the device it is being interviewed and reports success.
But then presence is frozen on null or true.
nobody's at home for an hour or so and:

[image]

[image]

[image]

[image]

Illuminance levels are far away from my knx sensors that report several thousands lux compared to 2-3 lx from tuya.
Any ideas what to do next?
It never worked for me as I bought it 2 or 3 weeks ago.
Do you think this could be faulty device?
I do not have tuya gateway to check with native app.