Koenkk/zigbee2mqtt #27064: MQTT error + Failed to Start 'HomeAssistant'

robertatc, 2025-04-10:
### What happened?

Hello everyone,

My configuration inside a MINI-PC with Promox:

1. HAOS
2. MQTT
3. Zigbee2MQTT

Everytime that I start (Because i need to restart the Zigbee), it shows me all the time the next error after few minutes:

`zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: MQTT error: Keepalive timeout
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Failed to start 'HomeAssistant' (Error: Connection closed
zigbee2mqtt  |     at /app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:1647:34
zigbee2mqtt  |     at Array.forEach (<anonymous>)
zigbee2mqtt  |     at MqttClient._flushVolatile (/app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:1642:31)
zigbee2mqtt  |     at Socket.<anonymous> (/app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:828:9)
zigbee2mqtt  |     at Socket.emit (node:events:524:28)
zigbee2mqtt  |     at TCP.<anonymous> (node:net:351:12))
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Not connected to MQTT server!
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Cannot send message: topic: 'zigbee2mqtt/Repetidor Planta Primera', payload: '{"device":{"applicationVersion":8,"dateCode":"20210703","friendlyName":"Repetidor Planta Primera","hardwareVersion":1,"ieeeAddr":"0x00124b00256033e8","manufacturerID":0,"manufacturerName":"eWeLink","model":"SA-003-Zigbee","networkAddress":8470,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.7","type":"Router","zclVersion":1},"linkquality":190}
zigbee2mqtt  | [2025-04-10 09:22:32] info:      z2m: Zigbee2MQTT started!
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Not connected to MQTT server!
zigbee2mqtt  | [2025-04-10 09:22:33] info:      z2m: Connected to MQTT server`

Doesn't care if I've a device or not, it gives the error if I've the homeassistant enabled true at the configuration.yaml.

I did pings, telnet between all the machines and no problem or disconnections between them, so the network is stable. I've VLANs but all of them are in the same. 

Thanks in advance ;) 

### What did you expect to happen?

No error from Zigbee2MQTT

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240705

### Adapter

SLZB-06P10

### Setup

HomeAssistant OS, VM of Promox
MQTT: Debian + Docker
Zigbee2MQTT: Debian + Docker

### Debug log

`zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: MQTT error: Keepalive timeout
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Failed to start 'HomeAssistant' (Error: Connection closed
zigbee2mqtt  |     at /app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:1647:34
zigbee2mqtt  |     at Array.forEach (<anonymous>)
zigbee2mqtt  |     at MqttClient._flushVolatile (/app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:1642:31)
zigbee2mqtt  |     at Socket.<anonymous> (/app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:828:9)
zigbee2mqtt  |     at Socket.emit (node:events:524:28)
zigbee2mqtt  |     at TCP.<anonymous> (node:net:351:12))
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Not connected to MQTT server!
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Cannot send message: topic: 'zigbee2mqtt/Repetidor Planta Primera', payload: '{"device":{"applicationVersion":8,"dateCode":"20210703","friendlyName":"Repetidor Planta Primera","hardwareVersion":1,"ieeeAddr":"0x00124b00256033e8","manufacturerID":0,"manufacturerName":"eWeLink","model":"SA-003-Zigbee","networkAddress":8470,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.7","type":"Router","zclVersion":1},"linkquality":190}
zigbee2mqtt  | [2025-04-10 09:22:32] info:      z2m: Zigbee2MQTT started!
zigbee2mqtt  | [2025-04-10 09:22:32] error:     z2m: Not connected to MQTT server!
zigbee2mqtt  | [2025-04-10 09:22:33] info:      z2m: Connected to MQTT server`

robertatc, 2025-04-10:
@Nerivec 
Thanks for the input. I test the ping and telnet during 1 hour an no errors so far. However, at Z2M it has some disconnections, it quiet strange.

Nerivec, 2025-04-10:
What MQTT broker are you using, and which version do you have configured in the `configuration.yaml`?

robertatc, 2025-04-11:
@Nerivec 

I'm using the **image: eclipse-mosquitto:latest**, and this is the information from the configuration.yaml (Using version 4 of Z2M).

```
version: 4
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.50.110:1883
  user: zigbee
  password: ************************
  keepalive: 60
  include_device_information: true
  reject_unauthorized: true
```

Thanks for the help

Nerivec, 2025-04-11:
> Using version 4 of Z2M

I meant the `mqtt` > `version` ([docs](https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#server-connection)), but that code block answered the question (since you don't having it, it's using the default: 4).

What kind of latency do you have when you ping the mqtt server?
Have you tried pushing the keepalive to 120 (or higher) maybe?
Any chance you have something in your network that could affect the line between the Z2M server and the MQTT server (I'm assuming you're pinging from a 3rd party, so it's not really Z2M<>MQTT)?

robertatc, 2025-04-11:
No latency at all, have a look at the screenshot. The ping is from same VLANs.

[image]

Yes I did, still the same. However, what I check now is that is happening only when I start Z2M. If I just keep look for the log during one hour more, nothing more so far.

Both machines are in the same VLAN, both of them are inside a VM of promox, the rest of the pings are okay, ssh, anything related to connect between both machines is good. Also, before posted the first issued, I reinstalled both machines just in case, but it was the same at the end.

What I tested is if I set homeassistant: enable false, there is no error so far. 

Thanks for the help Nerivec

Nerivec, 2025-04-11:
Can you grab some `debug` logs of the whole startup sequence until the error from the first post?

robertatc, 2025-04-15:
Sorry for the late reply.

This is what I've, is strange because before I have also the FrontEnd Started, and now is not. 
```
[2025-04-14 19:26:58] error:    z2m: MQTT error: Keepalive timeout
[2025-04-14 19:26:58] error:    z2m: Failed to start 'HomeAssistant' (Error: Connection closed
    at /app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:1647:34
    at Array.forEach (<anonymous>)
    at MqttClient._flushVolatile (/app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:1642:31)
    at Socket.<anonymous> (/app/node_modules/.pnpm/mqtt@5.10.4/node_modules/mqtt/src/lib/client.ts:828:9)
    at Socket.emit (node:events:524:28)
    at TCP.<anonymous> (node:net:351:12))
```

Nerivec, 2025-04-15:
Could be linked to https://github.com/Koenkk/zigbee2mqtt/issues/27125#issuecomment-2805818538
We need a full set of `debug` logs though (probably logs from HA too, to confirm where this is coming from).

closed 2025-05-01 by Nerivec