Koenkk/zigbee2mqtt #27067: TS0601_smart_CO_air_box (_TZE204_7bztmfm1) does expose carbon_monoxide binary wrong to Home Assistant

marcelhoogantink, 2025-04-10:
### What happened?

The sensor `carbon_monoxide` is exposed wrong to Home Assistant.
I added the woring defention:

Original definition:
```
    {
        fingerprint: tuya.fingerprint("TS0601", ["_TZE200_7bztmfm1", "_TZE204_7bztmfm1"]),
        model: "TS0601_smart_CO_air_box",
        vendor: "Tuya",
        description: "Smart air box (carbon monoxide)",
        fromZigbee: [legacy.fromZigbee.tuya_CO],
        toZigbee: [],
        exposes: [e.carbon_monoxide(), e.co()],
    },

```
Working defenition (changed `valueOn/valueOff` from `true/false` to `ON/OFF`):
```
{
        fingerprint: tuya.fingerprint("TS0601", ["_TZE200_7bztmfm1", "_TZE204_7bztmfm1"]),
        model: "TS0601_smart_CO_air_box",
        vendor: "Tuya",
        description: "Smart air box (carbon monoxide TEST)",
        fromZigbee: [legacy.fromZigbee.tuya_CO],
        toZigbee: [],
        exposes: [
			e.binary("carbon_monoxide", ea.STATE, "ON", "OFF").withDescription("Indicates if CO (carbon monoxide) is detected"),
			e.co()
		],
};

```

### What did you expect to happen?

See above

### How to reproduce it (minimal and precise)

See above

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20210708

### Adapter

 "zstack","port": "/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0"

### Setup

Add on Om Home assistant on Raspberry Pi 4b

Koenkk, 2025-04-10:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

marcelhoogantink, 2025-04-10:
Done

closed 2025-04-10 by marcelhoogantink