Koenkk/zigbee2mqtt #27068: [New device support]:

Liionboy, 2025-04-10:
### Link

https://daver.ro/produs/pachet-automatizare-premium-zigbee-clima-pentru-incalzire-si-racire-din-distribuitor-cu-6-zone-cu-pompa-de-caldura-daver/

### Database entry

{"id":47,"type":"EndDevice","ieeeAddr":"0x583bc2fffe03a71a","nwkAddr":37566,"manufId":4098,"manufName":"_TZE200_mzik0ov2","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_mzik0ov2","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1744279075919}

### Zigbee2MQTT version

2.2.1

### Comments

Datapoints:

{"16":"Current Temperature","49":"Gear","50":"Set temperature","101":"℃℉","102":"Heat","108":"Display Exttemp","109":"Temp Calibration","110":"Int Sensor Deadzone","114":"Max Temperature Limit","115":"Min Temperature Limit","117":"Device State On Power","118":"ProgramMode Type","119":"Programming - Workday Parameter 1","120":"Programming - Workday Parameter 2","121":"Programming - Rest Day Parameter 1","122":"Programming - Rest Day Parameter 2","125":"Power","126":"Set Temp","127":"Current Temperature:","128":"Mode","129":"Lock","130":"Fault"}

[image]

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_mzik0ov2',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Is a new Thermostat version **_TZE200_mzik0ov2** with a TS0601 , now only Linkquality is show after connect to Z2M, i tryed with a external convertor but is not work:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

module.exports = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_mzik0ov2',
        },
    ],
    model: 'TS0601_mzik0ov2',
    vendor: 'Tuya',
    description: 'Tuya TS0601 Thermostat with weekly schedule',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [],
    exposes: [
        e.temperature().withAccess(ea.STATE).withDescription('Current Temperature'),
        e.numeric('target_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(35)
            .withUnit('°C')
            .withDescription('Set temperature'),
        e.enum('mode', ea.STATE_SET)
            .withDescription('Operation mode')
            .withValues(['auto', 'manual', 'holiday']),
        e.binary('lock', ea.STATE_SET, 'LOCK', 'UNLOCK')
            .withDescription('Child lock'),
        e.binary('power', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Power state'),
    ],
    meta: {
        tuyaDatapoints: [
            [16, 'local_temperature', tuya.valueConverter.divideBy10],
            [126, 'target_temperature', tuya.valueConverter.divideBy10],
            [128, 'mode', tuya.valueConverterBasic.lookup({0: 'manual', 1: 'auto', 2: 'holiday'})],
            [129, 'lock', tuya.valueConverter.lockUnlock],
            [125, 'power', tuya.valueConverter.onOff],
        ],
    },
};`

Liionboy, 2025-04-11:
@Koenkk can you help me how to do a PR for supporting this new device? Or can you do one for me whit that external convertor? Because is work 100% percent .

Thanks

Koenkk, 2025-04-12:
@Liionboy please create a PR but make sure to use the `tuyaDatapoints` api instead (see `tuya.ts` for examples)

Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Liionboy, 2025-04-12:
@Koenkk this is the problem, i don't know how to do a PR, maybe you can help me with this, i don't know where and how to edit `tuya.js`

Koenkk, 2025-04-13:
@Liionboy what happens if you click that link?

Liionboy, 2025-04-13:
It open tuya.js but like i told you i don't know how much to edit that file like you say, this converter was made with help of Google AI Studio

Liionboy, 2025-04-14:
Hello @Koenkk 

Can you check if is ok like this?

```
{
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_mzik0ov2']),
    model: 'HY09RF-Zigbee',
    vendor: 'Tuya',
    description: 'Smart thermostat Daver(Cool/Heat/Schedule)',
    fromZigbee: [fzLocal.thermostat_tzeko],
    toZigbee: [tzLocal.thermostat_tzeko],
    exposes: [
        exposes.binary('power', ea.STATE_SET, 'ON', 'OFF').withDescription('Main power control (DP 125)'),
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 35, 0.5, ea.STATE_SET) // DP 50, Scalare 10
            .withLocalTemperature(ea.STATE) // DP 16, Scalare 10
            .withSystemMode(['off', 'cool', 'heat'], ea.STATE_SET) // DP 49 + DP 125
            .withRunningState(['idle', 'heat', 'cool'], ea.STATE) // DP 102 (Verifică valoare cool!)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET) // DP 109, Scalare 10
            .withPreset(['manual', 'program', 'holiday', 'temporary_manual'], ea.STATE_SET) // DP 128
            ,
        exposes.numeric('floor_temperature_sensor', ea.STATE).withUnit('°C').withDescription('Temperature reported by floor sensor (DP 103)'), // DP 103, Scalare 10? ReadOnly
        e.child_lock(), // DP 129, Bool?
        e.numeric('max_temperature_limit', ea.STATE_SET).withUnit('°C').withValueMin(20).withValueMax(70).withValueStep(1).withDescription('Maximum heating temperature limit setting'), // DP 114, Scalare 1? Verifică!
        e.numeric('min_temperature_limit', ea.STATE_SET).withUnit('°C').withValueMin(1).withValueMax(15).withValueStep(1).withDescription('Minimum heating temperature limit setting'), // DP 115, Scalare 1? Verifică!
        e.numeric('deadzone_temperature', ea.STATE_SET).withUnit('°C').withValueMin(0.5).withValueMax(2.5).withValueStep(0.1).withDescription('Temperature deadzone (hysteresis)'), // DP 110, Scalare 10
        e.enum('power_on_behavior', ea.STATE_SET, ['previous', 'off', 'on']).withDescription('State after power loss'), // DP 117, Enum Numeric (0,1,2)
        e.enum('temperature_unit', ea.STATE_SET, ['celsius', 'fahrenheit']).withDescription('Device temperature unit'), // DP 101, Bool (false/true)
        e.numeric('fault_code', ea.STATE).withDescription('Device fault code (0 = OK)'), // DP 130, Bitmap/Value
        e.enum('schedule_type', ea.STATE_SET, ['5+2', '6+1', '7']).withDescription('Schedule mode type (DP 118)'), // DP 118, Enum Numeric
        exposes.numeric('holiday_days', ea.STATE_SET).withValueMin(1).withValueMax(60).withValueStep(1).withUnit('days').withDescription('Duration for holiday mode (DP 104)'), // DP 104, Scalare 1 (Verifică Max!)
        exposes.numeric('holiday_temperature', ea.STATE_SET).withValueMin(5).withValueMax(30).withValueStep(1).withUnit('°C').withDescription('Target temperature for holiday mode (DP 105)'), // DP 105, Scalare 1 (Verifică Min/Max/Step!)
        // Expuneri text pentru program (citire și scriere)
        exposes.text('schedule_workday', ea.STATE_SET | ea.STATE).withDescription('Workday schedule (6 periods HH:MM/TT.T separated by space)'),
        exposes.text('schedule_restday', ea.STATE_SET | ea.STATE).withDescription('Restday schedule (6 periods HH:MM/TT.T separated by space)'),
    ],
    meta: {
        tuyaDatapoints: [
            [16, 'local_temperature', tuya.valueConverter.divideBy10],
            [50, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [101, 'temperature_unit', tuya.valueConverter.raw],
            [102, 'running_state', tuya.valueConverter.raw],
            [103, 'floor_temperature_sensor', tuya.valueConverter.divideBy10],
            [104, 'holiday_days', tuya.valueConverter.raw],
            [105, 'holiday_temperature', tuya.valueConverter.raw],
            [109, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.divideBy10],
            [114, 'max_temperature_limit', tuya.valueConverter.raw],
            [115, 'min_temperature_limit', tuya.valueConverter.raw],
            [117, 'power_on_behavior', tuya.valueConverter.raw],
            [118, 'schedule_type', tuya.valueConverter.raw],
            [119, 'schedule_workday_part1', tuya.valueConverter.raw],
            [120, 'schedule_workday_part2', tuya.valueConverter.raw],
            [121, 'schedule_restday_part1', tuya.valueConverter.raw],
            [122, 'schedule_restday_part2', tuya.valueConverter.raw],
            [125, 'system_mode', tuya.valueConverter.raw],
            [126, 'occupied_heating_setpoint', tuya.valueConverter.raw],
            [128, 'preset', tuya.valueConverter.raw],
            [129, 'child_lock', tuya.valueConverter.raw],
            [130, 'fault_code', tuya.valueConverter.raw],
            [49, 'system_mode', tuya.valueConverter.raw],
            [142, 'power', tuya.valueConverter.raw],
        ],
    },
}
```

Liionboy, 2025-04-14:
Hello @Koenkk 

Can you tell me if is ok?

```
{
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_mzik0ov2']),
    model: 'HY09RF-Zigbee',
    vendor: 'Tuya',
    description: 'Smart thermostat Daver (Cool/Heat/Schedule)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        exposes.binary('power', ea.STATE_SET, 'ON', 'OFF').withDescription('Main power control (DP 125)'), // VERIFICĂ! Tip Bool/Enum?
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSystemMode(['off', 'cool', 'heat'], ea.STATE_SET)
            .withRunningState(['idle', 'heat', 'cool'], ea.STATE) // VERIFICĂ! Val cool?
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withPreset(['manual', 'program', 'holiday', 'temporary_manual'], ea.STATE_SET)
            ,
        exposes.numeric('floor_sensor_temperature', ea.STATE).withUnit('°C').withDescription('Temperature reported by floor sensor (DP 103)'), // VERIFICĂ! Scalare 10? ReadOnly
        e.child_lock(), // VERIFICĂ! Bool?
        e.numeric('max_temperature_limit', ea.STATE_SET).withUnit('°C').withValueMin(15).withValueMax(45).withValueStep(1).withDescription('Maximum heating temperature limit setting (DP 114)'), // VERIFICĂ! Scalare 1? Limite?
        e.numeric('min_temperature_limit', ea.STATE_SET).withUnit('°C').withValueMin(1).withValueMax(15).withValueStep(1).withDescription('Minimum heating temperature limit setting (DP 115)'), // VERIFICĂ! Scalare 1? Limite?
        e.numeric('deadzone_temperature', ea.STATE_SET).withUnit('°C').withValueMin(0.5).withValueMax(2.5).withValueStep(0.1).withDescription('Temperature deadzone (hysteresis)'), // DP 110, Scalare 10
        e.enum('power_on_behavior', ea.STATE_SET, ['previous', 'off', 'on']).withDescription('State after power loss'),
        e.enum('temperature_unit', ea.STATE_SET, ['celsius', 'fahrenheit']).withDescription('Device temperature unit'),
        e.numeric('fault_code', ea.STATE).withDescription('Device fault code (0 = OK)'),
        e.enum('schedule_type', ea.STATE_SET, ['5+2', '6+1', '7']).withDescription('Schedule mode type (DP 118)'),
        exposes.numeric('holiday_days', ea.STATE_SET).withUnit('days').withValueMin(1).withValueMax(60).withValueStep(1).withDescription('Duration for holiday mode (DP 104)'), // VERIFICĂ Max!
        exposes.numeric('holiday_temperature', ea.STATE_SET).withUnit('°C').withValueMin(5).withValueMax(30).withValueStep(1).withDescription('Target temperature for holiday mode (DP 105)'), // VERIFICĂ Min/Max/Step!
        exposes.text('schedule_workday', ea.STATE_SET | ea.STATE).withDescription('Workday schedule (6 periods HH:MM/TT.T separated by space)'),
        exposes.text('schedule_restday', ea.STATE_SET | ea.STATE).withDescription('Restday schedule (6 periods HH:MM/TT.T separated by space)'),
    ],
    meta: {
        // Adăugăm mapările necesare pentru lookup-uri
        tuyaThermostatSystemModeDp49: { 0: 'cool', 1: 'heat' }, // VERIFICĂ! Map 0/1?
        tuyaThermostatPreset: { 0: 'manual', 1: 'program', 2: 'holiday', 3: 'temporary_manual' },
        tuyaThermostatPowerOnState: { 0: 'previous', 1: 'off', 2: 'on' },
        tuyaScheduleTypeMap: { 0: '5+2', 1: '6+1', 2: '7' },
        // Lista DP-urilor gestionate de convertoarele standard
        tuyaDatapoints: [
            // DP 16: Temp Curentă - necesită convertor custom mic în fz pt a ignora 0
            // {dp: 16, name: 'local_temperature', converter: tuya.valueConverter.divideBy10, cluster: 'manuSpecificTuya'}, // Varianta standard NU ignoră 0
            // DP 50: Setpoint Heat - OK standard
            {dp: 50, name: 'occupied_heating_setpoint', valueConverter: tuya.valueConverter.divideBy10}, // pt fz
            {dp: 50, name: 'occupied_heating_setpoint', valueConverter: tuya.valueConverter.multiplyBy10}, // pt tz
            // DP 109: Calibrare - OK standard
            {dp: 109, name: 'local_temperature_calibration', valueConverter: tuya.valueConverter.divideBy10},
            {dp: 109, name: 'local_temperature_calibration', valueConverter: tuya.valueConverter.multiplyBy10},
            // DP 114: Limită Max - OK standard (dacă scalare 1)
            {dp: 114, name: 'max_temperature_limit', valueConverter: tuya.valueConverter.raw}, // VERIFICĂ! Scalare 1?
            {dp: 114, name: 'max_temperature_limit', valueConverter: tuya.valueConverter.raw}, // VERIFICĂ! Scalare 1?
            // DP 115: Limită Min - OK standard (dacă scalare 1)
            {dp: 115, name: 'min_temperature_limit', valueConverter: tuya.valueConverter.raw}, // VERIFICĂ! Scalare 1?
            {dp: 115, name: 'min_temperature_limit', valueConverter: tuya.valueConverter.raw}, // VERIFICĂ! Scalare 1?
            // DP 110: Deadzone - OK standard
            {dp: 110, name: 'deadzone_temperature', valueConverter: tuya.valueConverter.divideBy10},
            {dp: 110, name: 'deadzone_temperature', valueConverter: tuya.valueConverter.multiplyBy10},
            // DP 129: Child Lock - OK standard (dacă e Bool)
            {dp: 129, name: 'child_lock', valueConverter: tuya.valueConverter.lockUnlock}, // VERIFICĂ! Bool?
            // DP 101: Unitate Temp - OK standard
            {dp: 101, name: 'temperature_unit', valueConverter: tuya.valueConverter.celsiusFahrenheit},
            // DP 117: Power On Behavior - OK standard cu lookup
[... cut ...]
            {dp: 105, name: 'holiday_temperature', valueConverter: tuya.valueConverter.raw, meta: { min: 5, max: 30 }}, // VERIFICĂ MIN/MAX!
            // DP 103: Temp Podea - OK standard (dacă scalare 10)
            {dp: 103, name: 'floor_sensor_temperature', valueConverter: tuya.valueConverter.divideBy10}, // VERIFICĂ! Scalare 10? ReadOnly in exposes
            // DP 49, 102, 125 sunt gestionate de fz_local/tz_local
            // DP 119-122 sunt gestionate de fz_local/tz_local
        ],
    },
}
```

Koenkk, 2025-04-14:
> Can you tell me if is ok?

It doesn't look ok to me, I guess it doesn't work when testing it?

Liionboy, 2025-04-14:
> > Can you tell me if is ok?
> 
> It doesn't look ok to me, I guess it doesn't work when testing it?

For me is work with External Converter, so maybe someone want to do a PR for me is ok to use with Converter.

Thanks

Liionboy, 2025-04-16:
Hello @Koenkk ,

I did another convertor file, can you check if now i am using like you say in forst comment?

But still i am using some custom function, please if you have time check a little bit this external convertor and tell me what part form him need to do a PR.

```
// Ultima actualizare: Corectat mapare DP49 (cool=1, heat=0)

// --- Importuri ---
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ea = exposes.access;
const e = exposes.presets;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const { precisionRound } = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const Buffer = require('buffer').Buffer;

// --- Funcții Ajutătoare pentru Program ---
const parseScheduleData = (buffer) => {
    if (!Buffer.isBuffer(buffer) || buffer.length === 0 || buffer.length % 3 !== 0) return null;
    const periods = [];
    try {
        for (let i = 0; i < buffer.length; i += 3) {
            const hour = buffer.readUInt8(i), minute = buffer.readUInt8(i + 1), tempRaw = buffer.readUInt8(i + 2);
            const tempCelsius = precisionRound(tempRaw / 1, 1); // Scalare /1 conform original
            if (hour > 23 || minute > 59) continue; // Skip invalid time
            const timeStr = `${hour.toString().padStart(2, '0')}:${minute.toString().padStart(2, '0')}`;
            periods.push(`${timeStr}/${tempCelsius.toFixed(1)}°C`);
        }
        return periods.length > 0 ? periods.join(' ') : null;
    } catch (e) { return null; } // Return null on error
};

const parseInputScheduleString = (scheduleString, expectedPeriods) => {
    if (typeof scheduleString !== 'string') return null; // Check input type
    const periods = scheduleString.trim().split(/\s+/);
    if (periods.length === 1 && periods[0] === '') periods.shift(); // Handle empty string
    if (periods.length !== expectedPeriods) return null; // Validate number of periods

    const dataBytes = [];
    for (const period of periods) {
        const parts = period.split('/'); if (parts.length !== 2) return null; // HH:MM/TT.T format check
        const timeParts = parts[0].split(':'); if (timeParts.length !== 2) return null; // HH:MM format check
        const hour = parseInt(timeParts[0], 10), minute = parseInt(timeParts[1], 10);
        const tempStr = parts[1].replace(/°C/i, ''), tempCelsius = parseFloat(tempStr);
        // Validate parsed values
        if (isNaN(hour) || hour < 0 || hour > 23 || isNaN(minute) || minute < 0 || minute > 59 || isNaN(tempCelsius)) return null;
        const tempRaw = Math.round(tempCelsius * 1); // Scalare *1 conform original
        if (tempRaw < 0 || tempRaw > 255) return null; // Check raw temp bounds (1 byte)
        dataBytes.push(hour, minute, tempRaw); // Add bytes for the period
    }
    return Buffer.from(dataBytes); // Return buffer
};
const totalPeriodsPerDay = 6; // Define globally

// --- Convertor Custom DOAR PENTRU CITIRE Program ---
const fzScheduleConverter = {
    from: (value, meta, options, publish) => {
        // Handle potential initial call before meta.state is fully populated
        if (!meta || !meta.message || typeof meta.message.dp === 'undefined') {
             return parseScheduleData(value); // Attempt to parse current value anyway
        }
        if (!meta.state) meta.state = {}; // Initialize meta.state if it doesn't exist

        const dp = meta.message.dp;
        // Determine if it's workday/restday based on DP ID
        const isWorkday = (dp === 119 || dp === 120);
[... 240 lines cut ...]
            [103,   'floor_temperature_sensor',  tuya.valueConverter.divideBy10], // ReadOnly sensor
            [129,   'child_lock',                tuya.valueConverter.lockUnlock],
            [114,   'max_temperature_limit',     tuya.valueConverter.raw], // Assuming no scaling needed
            [115,   'min_temperature_limit',     tuya.valueConverter.raw], // Assuming no scaling needed
            [110,   'deadzone_temperature',      tuya.valueConverter.divideBy10],
            [117,   'power_on_behavior',         tuya.valueConverterBasic.lookup({'previous': 0, 'off': 1, 'on': 2})],
            [101,   'temperature_unit',          tuyaTempUnitConverter],       // Uses custom boolean converter
            [130,   'fault_code',                tuya.valueConverter.raw],     // ReadOnly fault code
            [118,   'schedule_type',             tuya.valueConverterBasic.lookup({'5+2': 0, '6+1': 1, '7': 2})],
            // Holiday DPs
            [104,   'holiday_days',              tuya.valueConverter.raw], // Assuming no scaling
            [105,   'holiday_temperature',       tuya.valueConverter.raw], // Assuming no scaling
            // Schedule DPs (Reading handled by fzScheduleConverter)
            // Setting is handled by tzLocal_schedule
            [119,   'schedule_workday',          fzScheduleConverter],       // Part 1 Workday
            [121,   'schedule_restday',          fzScheduleConverter],       // Part 1 Restday
            // Define Part 2 DPs so fzScheduleConverter receives their data too
            [120,   null,                        fzScheduleConverter],       // Part 2 Workday (associated with fzScheduleConverter)
            [122,   null,                        fzScheduleConverter],       // Part 2 Restday (associated with fzScheduleConverter)
         ],
     },
};

module.exports = definition; // Export the definition for Zigbee2MQTT
```

closed 2025-06-23 by github-actions[bot]