Koenkk/zigbee2mqtt #27077: Z2M replaces  "'!/config/secrets.yaml mqtt_password'" with the actual password

asolochek, 2025-04-11:
### What happened?

I keep my home assistant configuration, including my zigbee2mqtt/configuration.yaml in git. Because of this, I use the !secrets substitution in that file:
```
  server: '!/config/secrets.yaml mqtt_server'
  user: '!/config/secrets.yaml mqtt_user'
  password: '!/config/secrets.yaml mqtt_password'
```

I noticed while checking in some unrelated changes that my configuration.yaml was marked as modified, and I saw that those three fields were replaced with the values from the secrets.yaml file. I reverted that change and restarted zigbee2mqtt, and saw once again configuration.yaml was modified. 

### What did you expect to happen?

I expected that configuration.yaml would not be modified unless I changed something in my zigbee2mqtt configuration. 

### How to reproduce it (minimal and precise)

create /config/secrets.yaml with contents like:
```
mqtt_server: mqtt://your-server:1833
mqtt_user: username
mqtt_password: yourpassword
```
and modify /config/zigbee2mqtt/configuration.yaml so these fields are set thusly:
```
  server: '!/config/secrets.yaml mqtt_server'
  user: '!/config/secrets.yaml mqtt_user'
  password: '!/config/secrets.yaml mqtt_password'
```
restart zigbee2mqtt and observe that configuration.yaml has been updated to contain:
```
  server: mqtt://your-server:1883
  user: username
  password: yourpassword
```

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

2.5.6

### Adapter

SLZB-06

### Setup

Add-on on HAOS running in a libvirt VM.

Nerivec, 2025-04-11:
Do you have settings in the HA add-on configuration page that could interfere (they take precedence over the `configuration.yaml`)?

asolochek, 2025-04-22:
In the z2m config? Not that I see.

```data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: tcp://XXXX:6638
  baudrate: 460800
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20}
```

asolochek, 2025-04-22:
The add-on config now looks identical to what you suggested. The configuration.yaml file actually already contained the serial config. Restarting z2m still rewrote configuration.yaml replacing the secrets.yaml references. I have also tried using "!secrets mqtt_server" without the .yaml and using double vs. single quotes, but that has made no difference.

Koenkk, 2025-04-27:
I cannot reproduce this issue with the `2.2.1-1` addon.

Addon config:

```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial: {}
```

`configuration.yaml`

```yaml
  GNU nano 8.2                                                      configuration.yaml                                                                 
version: 4
mqtt:
  base_topic: zigbee2mqtt_test
  server: '!secrets.yaml mqtt_server'
  user: '!secrets.yaml mqtt_user'
  password: '!secrets.yaml mqtt_password'
advanced:
  log_level: debug
...
```

`secrets.yaml`

```yaml
mqtt_server: mqtt://192.168.2.99
mqtt_user: user
mqtt_password: my_pass
```

asolochek, 2025-04-28:
Not sure why people can't reproduce it. My configuration is here: https://github.com/asolochek/HA-config/blob/main/zigbee2mqtt/configuration.yaml if you want to eliminate that as a variable (assuming you have some test install that won't get all messed up by introducing a bunch of new z2m devices).

ryan-fessler, 2025-05-18:
I also am running into this same bug on zigbee2mqtt version 2.3.0-1. Fields "server", "user", and "password" all get replaced with the value from secrets.yaml. However, "network_key" does not and correctly stays set to '!secrets.yaml network_key'.

h3llrais3r, 2025-05-22:
> I cannot reproduce this issue with the `2.2.1-1` addon.
> 

Can you reproduce it with the latest ´2.3.0-1´ addon version?

Koenkk, 2025-06-01:
@h3llrais3r it seems when you put it in the add-on config it is taken from the Home Assistant `configuration.yaml`:

[image]

I guess Home Assistant then replaces e.g. `"!secret mqtt_server"` with the actual value before passing it to the Z2M add-on. Can you try removing it from the add-on config page?

h3llrais3r, 2025-06-01:
I finally got it working.
This is my addon config:
```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: "!secret.yaml mqtt_server"
  user: "!secret.yaml mqtt_user"
  password: "!secret.yaml mqtt_password"
serial:
  adapter: zstack
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
watchdog: default
```

This is what I've put in the `configuration.xml`:

[image]

What's the difference?
I was using `!secret ...` in addon config and a mix of `!secret ...` and `!secret.yaml ...` in the configuration xml.
I've now alligned all secrets as `!secret.yaml ...` and now it starts properly, without replacing anything inside the configuration.xml

asolochek, 2025-06-02:
> What's the difference? I was using `!secret ...` in addon config and a mix of `!secret ...` and `!secret.yaml ...` in the configuration xml. I've now alligned all secrets as `!secret.yaml ...` and now it starts properly, without replacing anything inside the configuration.xml
> 
> The addon is probably reading the values from my home assistant `secrets.xml` and is replacing the real value in the configuration.xml. When I'm using the exact same notation in addon and configuration.xml, it doesn't replace anything, but it's still properly replaced at runtime.

First, I'm assuming you mean .yaml everywhere you just wrote .xml, right? 

If you remove the !secrets from your add-on config entirely, so the mqtt section is just like this:

```
mqtt: {}
```

Does that still work for you, or does it cause your configuration.yaml to get modified again?

h3llrais3r, 2025-06-02:
Sorry, it should have been  `yaml` instead of `xml`.
I've edited my previous posts.

If I remove the mqtt config from the addon, the addon starts with the wrong user (`addons` instead of my custom username) and writes those wrong values to the zigbee2mqtt configuration.yaml. (see my second previous post)
So the solution for me is using the exact same secret notation (`!secret.yaml <secret_name>`) in both addon config as zigbee2mqtt configuration.yaml. (see previous post)

asolochek, 2025-06-02:
Ok, I added the secrets to the addon config exactly as you had it, and removed server, user, and password from configuration.yaml. When I restarted the addon it added these lines back:

```
  server: '!secret.yaml mqtt_server'
  user: '!secret.yaml mqtt_user'
  password: '!secret.yaml mqtt_password'
```
So while there is still some strange bug here with the addon rewriting the configuration.yaml file in general, and having to have those lines in both the configuration.yaml file and in the addon config is annoying, this is still a vastly improved situation since I no longer have my password getting saved to a file tracked on github.

ryan-fessler, 2025-06-08:
I can confirm that having those lines in both configuration.yaml and the addon config fixes the problem.

closed 2025-08-16 by github-actions[bot]