Koenkk/zigbee2mqtt #27079: Devices not showing up in Zigbee2MQTT

Lonewolfazer, 2025-04-11:
### What happened?

I’m reaching out for help regarding an issue I’ve been experiencing with device discovery and visibility in the Zigbee2MQTT frontend.
I recently migrated my setup from Jeedom to Home Assistant. I’m using a ConBee III stick with the `deconz` adapter, and Zigbee2MQTT is running in a Home Assistant add-on environment. My broker (Mosquitto) is working fine and Home Assistant is successfully connected to it.

What I’ve tried:
- Restored the original `database.db` and `state.json` files.
- Validated my `configuration.yaml` (Z2M starts without critical errors).
- Replaced and tested multiple versions of the `database.db` file (from backups).
- Verified that my MQTT broker is functional (no connection errors in logs).
- Deleted `database.db` and let Z2M recreate it — only the coordinator appears.
- Tried switching between `deconz` and `ezsp` (went back to `deconz` which used to work fine).
- Zigbee2MQTT logs mention "updates available", so the devices seem known somehow — but still invisible in the UI.

### What did you expect to happen?

Despite having a `database.db` and `state.json` file (extracted from my Jeedom setup), **no devices are showing up in the Zigbee2MQTT frontend**.  
However, some Zigbee devices are still functional via Home Assistant (e.g., I can turn on a light), which means they are connected and responding on the Zigbee network.

Strange behavior:
Some devices (e.g., `0x540f57fffe44a8b6`) are listed in `state.json` and `database.db`, and even appear in HA via MQTT entities — but Z2M doesn’t list them in its Devices tab, and I cannot interact with them from there.

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

Coordinator firmware version: 0x26780700

### Adapter

combee III

### Setup

Add on Home Assistant OS 

### Debug log

[log.log](https://github.com/user-attachments/files/19701332/log.log)

Nerivec, 2025-04-11:
The problem should be fixed in dev/edge (caused by [missing check in converters](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9130) for Nodon devices). You can try switching to confirm: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

closed 2025-04-11 by Nerivec

Lonewolfazer, 2025-04-12:
That's good, it's working now with Edge