Koenkk/zigbee2mqtt #27080: Sonoff TRVZB | z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}

onurece, 2025-04-11:
### What happened?

Got the errors below at random times after receiving unexpected messages from Sonoff TRVZB.

`z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}`
`z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '3'}`

### What did you expect to happen?

Not receive unexpected `temperatureSensorSelect` value, i.e. 2 and 3, from devices or process all values of `temperatureSensorSelect` without errors.

### How to reproduce it (minimal and precise)

I haven't been able to pinpoint when Z2M receives a "wrong" message or what makes the devices submit a "wrong" message, but this error happens when a message with unexpected value is received. Sometimes only `temperatureSensorSelect` is included in the message. Sometimes both `temperatureSensorSelect` and `externalTemperatureInput` are included. I don't know why there is such a difference.

Here are some examples:
`[2025-04-10 23:38:34] debug: 	z2m: Received Zigbee message from 'Sonoff Living Room Horizontal TRV', type 'attributeReport', cluster 'customSonoffTrvzb', data '{"temperatureSensorSelect":2}' from endpoint 1 with groupID 0`
`[2025-04-10 23:38:39] debug: 	z2m: Received Zigbee message from 'Sonoff Living Room Horizontal TRV', type 'attributeReport', cluster 'customSonoffTrvzb', data '{"externalTemperatureInput":2140,"temperatureSensorSelect":3}' from endpoint 1 with groupID 0`

### Zigbee2MQTT version

2.2.1 commit: unknown

### Adapter firmware version

20240710

### Adapter

zstack

### Setup

Bare on RPi 5 

### Debug log

[log.log](https://github.com/user-attachments/files/19704926/log.log)

```
[2025-04-10 23:38:34] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,5,130,1,172,14,160,254,27,68,129,0,0,17,252,5,130,1,1,0,91,0,109,29,177,0,0,7,24,59,10,14,96,32,2,172,14,28,242]
[2025-04-10 23:38:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,5,130,1,172,14,160,254,27,68,129,0,0,17,252,5,130,1,1,0,91,0,109,29,177,0,0,7,24,59,10,14,96,32,2,172,14,28,242]
[2025-04-10 23:38:34] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [5,130,1,172,14] - 160
[2025-04-10 23:38:34] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-04-10 23:38:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,17,252,5,130,1,1,0,91,0,109,29,177,0,0,7,24,59,10,14,96,32,2,172,14,28,242]
[2025-04-10 23:38:34] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,17,252,5,130,1,1,0,91,0,109,29,177,0,0,7,24,59,10,14,96,32,2,172,14,28] - 242
[2025-04-10 23:38:34] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":64529,"srcaddr":33285,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":91,"securityuse":0,"timestamp":11607405,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,59,10,14,96,32,2]}}
[2025-04-10 23:38:34] debug: 	zh:controller: Received payload: clusterID=64529, address=33285, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=91, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":59,"commandIdentifier":10},"payload":[{"attrId":24590,"dataType":32,"attrData":2}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-04-10 23:38:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-04-10 23:38:34] debug: 	z2m: Received Zigbee message from 'Sonoff Living Room Horizontal TRV', type 'attributeReport', cluster 'customSonoffTrvzb', data '{"temperatureSensorSelect":2}' from endpoint 1 with groupID 0
[2025-04-10 23:38:34] error: 	z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}
[2025-04-10 23:38:34] debug: 	z2m: Error: Expected one of: 0, 1, got: '2'
    at getFromLookupByValue (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:678:15)
    at Object.convert (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2331:110)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:170:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)
```

Nerivec, 2025-04-11:
It appears you may have a version with a firmware that Z2M does not support (yet).
What's strange is that `temperature_sensor_select` is supposed to be `internal` or `external`, so not sure why you'd get values higher than 1 (`internal: 0`, `external: 1`), except maybe a firmware bug (using 2,3 instead of 0,1...).
What firmware build date and version do you have for that device?

onurece, 2025-04-11:
> It appears you may have a version with a firmware that Z2M does not support (yet). What's strange is that `temperature_sensor_select` is supposed to be `internal` or `external`, so not sure why [...]

I'm on the latest firmware for TRVZB, that is 1.3.0. Also, this feature became with version 1.3.0.

Ohh, I think I have an idea what's happening. If the temperature of the external sensor doesn't send any new data, e.g. if the temperature doesn't change for 2 hours, the TRV starts reporting the internal temperature as the current temperature. I believe the setting still stays as `external`, but it's ignored. I think this a "feature" not a bug to prevent the TRV getting stuck on a temperature because maybe the sensor is broken or lost connection, and it kind of makes sense, but two hours is too short.

In one of the cases, this weird message is sent at 2025-04-11 17:18:17.
```
[2025-04-11 17:18:17] debug: 	z2m: Received Zigbee message from 'Sonoff Guest Room TRV', type 'attributeReport', cluster 'customSonoffTrvzb', data '{"temperatureSensorSelect":2}' from endpoint 1 with groupID 0
[2025-04-11 17:18:17] error: 	z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}
[2025-04-11 17:18:17] debug: 	z2m: Error: Expected one of: 0, 1, got: '2'
    at getFromLookupByValue (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:678:15)
    at Object.convert (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2331:110)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:170:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)
```

This happens to be the exact time that the current temperature deviates from the temperature of the external sensor.
[image]

Then at 2025-04-11 18:07:00, I receive a new temperature reading and an expected value of `temperatureSensorSelect`, and the current temperature of the TRV becomes the same as what's reported by the external sensor.

```
[2025-04-11 18:07:00] debug: 	z2m: Received Zigbee message from 'Sonoff Guest Room TRV', type 'attributeReport', cluster 'customSonoffTrvzb', data '{"temperatureSensorSelect":1}' from endpoint 1 with groupID 0
[2025-04-11 18:07:01] debug: 	z2m: Received Zigbee message from 'Sonoff Guest Room TRV', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2040}' from endpoint 1 with groupID 0
```

[image]

This only explains the value of 2. 3 is also related to this, but I couldn't exactly figure out when it's received.

I guess the solution should be to understand what 2 and 3 exactly mean and possibly add them as options here

[image]

Koenkk, 2025-04-13:
@lcheng33775823 could you check this? What does value 2 or 3 mean for `temperatureSensorSelect`?

onurece, 2025-06-14:
@Koenkk  Is there any other way we can learn what they mean?

Koenkk, 2025-06-14:
@onurece is the temperature of your thermostat measured by an internal or external sensor?

onurece, 2025-06-21:
@Koenkk It’s external

Koenkk, 2025-06-21:
thanks, fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-06-21 by Koenkk