Koenkk/zigbee2mqtt #27083: Addon not starting: Port is not open

TheArczi, 2025-04-11:
### What happened?

Hello, today i have installed HA, when i try start zigbee2mqtt I have problem.

I tried with `/dev/ttyACM0` but nothing changed.
configuration:
`version: 4
mqtt:
  base_topic: zigbee2mqtt
  server: 
  user: xxx
  password: xxx
serial:
  adapter: zstack
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000143F3E1-if00
  baudrate: 115200
  rtscts: false
advanced:
  log_level: info
  channel: 25
  network_key:
    - 102
    - 24
    - 181
    - 65
    - 169
    - 20
    - 155
    - 175
    - 122
    - 52
    - 142
    - 134
    - 207
    - 226
    - 117
    - 154
  pan_id: 30201
  ext_pan_id:
    - 195
    - 96
    - 142
    - 218
    - 97
    - 51
    - 169
    - 65
frontend:
  enabled: true
  port: 8099
homeassistant:
  enabled: true
`

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

20250404.0

### Adapter

Texas_Instruments_TI_CC2531

### Setup

Generic x86-64 on DELL Wyse

### Debug log

[21:48:41] INFO: Preparing to start...
[21:48:41] INFO: Socat not enabled
[21:48:41] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-04-11 21:48:43] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-11 21:48:43] info: 	z2m: Starting Zigbee2MQTT version 2.2.1 (commit #unknown)
[2025-04-11 21:48:43] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-11 21:48:43] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000143F3E1-if00","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-04-11 21:48:43] info: 	zh:zstack:znp: Serialport opened
[2025-04-11 21:48:43] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-04-11 21:48:44] info: 	zh:zstack:znp: Port closed
[2025-04-11 21:48:44] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-04-11 21:48:44] error: 	z2m: Error while starting zigbee-herdsman
[2025-04-11 21:48:44] error: 	z2m: Failed to start zigbee-herdsman
[2025-04-11 21:48:44] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-04-11 21:48:44] error: 	z2m: Exiting...
[2025-04-11 21:48:44] error: 	z2m: Error: Port is not open
    at SerialPort.set (/app/node_modules/.pnpm/@serialport+stream@13.0.0/node_modules/@serialport/stream/dist/index.js:263:37)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/serialPort.ts:54:18
    at new Promise (<anonymous>)
    at SerialPort.asyncSet (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/serialPort.ts:53:22)
    at Znp.skipBootloader (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:182:43)
    at Znp.openSerialPort (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:116:13)
    at Znp.open (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:94:85)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:104:9)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)

Nerivec, 2025-04-11:
Looks like the connection to the adapter gets closed as soon as Z2M tries to talk to it.
Can you try:
- a different port
- a different extension cable
- re-flashing the firmware
- flashing a different firmware version

Where did you get that firmware 20250404.0?
Also: https://www.zigbee2mqtt.io/guide/adapters/zstack.html#not-recommended-cc2530-cc2531-cc2538

TheArczi, 2025-04-11:
- i was trying on all ports
- not use extension cable

A cannot re-flash or flash new firmware because i buy this device.

Sorry for bad `Adapter firmware version` version i have thought about ha version - where to fine adatper firmware version?

I was used HA on proxmox since the year but i'm not doing updated few month and after i do all of them at one i was have this error so i was change to Generic x86-64 because i was thinking about that some time. Ale problem z Zigbee2MQTT  still exist.

Koenkk, 2025-04-13:
@TheArczi did you disable ZHA? Point 4 of https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#error-srsp-sys-ping-after-6000ms-for-zstack-or-host-fatal-error-for-emberznet

TheArczi, 2025-04-14:
I never installed ZHA.

Koenkk, 2025-04-14:
I believe it is enabled by default and you need to explicitly disable it. 

If you cannot get it working, I would try to see if the adapter works when running [bare metal](https://www.zigbee2mqtt.io/guide/installation/01_linux.html) (to confirm the adapter is OK)