Koenkk/zigbee2mqtt #27086: [Feature request]: Make Shelly Sensor Addon Attributes available in Zigbee2Mqtt

JohannesNeu, 2025-04-12:
### Is your feature request related to a problem? Please describe

I have attached a Shelly 1PM Gen4 device to my Zigbee network and can see the Shelly measurements and toggle the switch.
After attaching the sensor addon to the Shelly and connecting a DS18B20, I can see the temperature in the WebUI of the shelly device, but it is not forwarded to the Zigbee network.

### Describe the solution you'd like

I would be happy if it were possible for zigbee2mqtt to interpret external DS18B20 temperatures provided by Shelly devices.

### Describe alternatives you've considered

An alternative would be not to use Zigbee at all and stick with Wifi.

### Additional context

If there is any information I can provide to make development easier, please tell me how to do so.

rionshin, 2025-04-13:
You need to ask Shelly do they expose the add on measures in Zigbee . I have some doubts they don’t do it

teodor-hristov, 2025-07-03:
> You need to ask Shelly do they expose the add on measures in Zigbee . I have some doubts they don’t do it

Yes, they don't but they can stream data to mqtt.