Koenkk/zigbee2mqtt #27090: _TZ3000_xkap8wtb

rodionos, 2025-04-12:
### What happened?

- Zigbee2MQTT version: 2.1.3
- Coordinator type: EZSP v8
- Coordinator revision: 6.10.3.0 build 297
- Coordinator IEEE Address: 0x7cc6b6fffe490f0f
- Frontend version: 0.9.4
- zigbee-herdsman-converters version: 23.2.0
- zigbee-herdsman version: 3.2.7

The reported values for `current` and `energy` attributes ignore divisors as covered in #13929

```
[2025-04-12 13:49:43.913] debug: 	z2m: Received Zigbee message from '0xa4c138891590a111', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
[2025-04-12 13:49:43.914] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-usb/0xa4c138891590a111', payload '{"current":0,"energy":5,"last_seen":"2025-04-12T10:49:43.912Z","linkquality":172,"power":0,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":213}'
[2025-04-12 13:49:44.315] debug: 	z2m: Received Zigbee message from '0xa4c138891590a111', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":14,"rmsCurrent":70,"rmsVoltage":213}' from endpoint 1 with groupID 0
[2025-04-12 13:49:44.319] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-usb/0xa4c138891590a111', payload '{"current":70,"energy":5,"last_seen":"2025-04-12T10:49:44.315Z","linkquality":180,"power":14,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":213}'
```

database.db

```json
{
    "id": 57,
    "type": "Router",
    "ieeeAddr": "0xa4c138891590a111",
    "nwkAddr": 53086,
    "manufId": 4417,
    "manufName": "_TZ3000_xkap8wtb",
    "powerSource": "Mains (single phase)",
    "modelId": "TS0001",
    "epList": [
        1,
        242
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 256,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                1794,
                2820,
                57344,
                57345
            ],
            "outClusterList": [
                10,
                25
            ],
            "clusters": {
                "57344": {
                    "attributes": {
                        "53251": "AAAA"
                    }
                },
                "genBasic": {
                    "attributes": {
                        "modelId": "TS0001",
                        "manufacturerName": "_TZ3000_xkap8wtb",
                        "powerSource": 1,
                        "zclVersion": 3,
                        "appVersion": 162,
                        "stackVersion": 2,
                        "hwVersion": 1,
                        "swBuildId": "0122052017"
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "onOff": 1,
                        "moesStartUpOnOff": 1,
                        "startUpOnOff": 1
                    }
                },
                "manuSpecificTuya_3": {
                    "attributes": {
                        "switchType": 0
                    }
                },
                "haElectricalMeasurement": {
                    "attributes": {
                        "rmsVoltage": 212,
                        "rmsCurrent": 0,
                        "activePower": 0
                    }
                },
                "seMetering": {
                    "attributes": {
                        "currentSummDelivered": 4
                    }
                }
            },
            "binds": [
                {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x7cc6b6fffe490f0f",
                    "endpointID": 1
                },
                {
                    "cluster": 2820,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x7cc6b6fffe490f0f",
                    "endpointID": 1
                },
                {
                    "cluster": 1794,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x7cc6b6fffe490f0f",
                    "endpointID": 1
                }
            ],
            "configuredReportings": [],
            "meta": {}
        },
        "242": {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList": [],
            "outClusterList": [
                33
            ],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 162,
    "stackVersion": 2,
    "hwVersion": 1,
    "swBuildId": "0122052017",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1744453063199
}
```

Not sure if the below error is related:

```
[2025-04-12 13:20:52.791] error: 	z2m: Failed to configure '0xa4c138891590a111', attempt 2 (Error: ZCL command 0xa4c138891590a111/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9)
    at Object.rmsVoltage (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:219:5)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3902:13)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

### What did you expect to happen?

Correct values for `current` and `energy` given that the device matched the fingerprint in [tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/84c0e9429e1f6249470d11911d1cea499f6cf4a4/src/devices/tuya.ts#L3856)

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Docker container

rodionos, 2025-04-12:
I upgraded my z2m instance to 2.2.1 and EmberZNet 7.4.5. The issue is still there.

rodionos, 2025-04-15:
This type of relay does not support `acCurrentDivisor` and `acCurrentMultiplier ` attributes which are set in the configuration method:

```typescript
            endpoint.saveClusterAttributeKeyValue("haElectricalMeasurement", {
                acCurrentDivisor: 1000,
                acCurrentMultiplier: 1,
            });
            endpoint.saveClusterAttributeKeyValue("seMetering", {
                divisor: 100,
                multiplier: 1,
            });
```

```
[2025-04-15 10:27:56.512] debug: 	zh:controller:endpoint: Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.read(["acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
[2025-04-15 10:27:56.513] error: 	z2m: Publish 'set' 'read' to '0xa4c138132936ae0e' failed: 'Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.read(["acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-04-15 10:27:56.514] debug: 	z2m: Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.read(["acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
[2025-04-15 10:27:58.157] warning: 	z2m: Failed to ping '0x403059fffec7cf11' (attempt 1/1, ZCL command 0x403059fffec7cf11/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '13831'.))
[2025-04-15 10:28:02.049] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt-usb/0xa4c138132936ae0e/1/set' with data '{"write":{"cluster":"haElectricalMeasurement","options":{},"payload":{"acCurrentMultiplier":1}}}'
[2025-04-15 10:28:02.050] debug: 	z2m: Publishing 'set' 'write' to '0xa4c138132936ae0e'
[2025-04-15 10:28:02.051] debug: 	zh:controller:endpoint: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.write({"acCurrentMultiplier":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-04-15 10:28:02.052] debug: 	zh:ember: ~~~> [ZCL to=0xa4c138132936ae0e:50821 apsFrame={"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":159,"commandIdentifier":2}]
[2025-04-15 10:28:02.090] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-usb/0xa4c138132936ae0e', payload '{"current":0,"energy":2,"last_seen":"2025-04-15T07:28:02.089Z","linkquality":128,"power":0,"power_outage_memory":"on","state":"OFF","switch_type":"toggle","voltage":217}'
[2025-04-15 10:28:02.093] debug: 	zh:controller:endpoint: Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.write({"acCurrentMultiplier":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
[2025-04-15 10:28:02.094] error: 	z2m: Publish 'set' 'write' to '0xa4c138132936ae0e' failed: 'Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.write({"acCurrentMultiplier":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-04-15 10:28:02.094] debug: 	z2m: Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.write({"acCurrentMultiplier":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
[2025-04-15 10:28:13.610] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt-usb/0xa4c138132936ae0e/1/set' with data '{"write":{"cluster":"haElectricalMeasurement","options":{},"payload":{"acCurrentDivisor":1000}}}'
[2025-04-15 10:28:13.611] debug: 	z2m: Publishing 'set' 'write' to '0xa4c138132936ae0e'
[2025-04-15 10:28:13.612] debug: 	zh:controller:endpoint: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.write({"acCurrentDivisor":1000}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-04-15 10:28:13.613] debug: 	zh:ember: ~~~> [ZCL to=0xa4c138132936ae0e:50821 apsFrame={"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":160,"commandIdentifier":2}]
[2025-04-15 10:28:13.643] debug: 	zh:ember:ezsp: ezspIncomingRouteRecordHandler: source=50821 sourceEui=0xa4c138132936ae0e lastHopLqi=128 lastHopRssi=-68 relayCount=0 relayList=
[... cut ...] 00,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-04-15 10:28:13.657] debug: 	z2m: Error: ZCL command 0xa4c138132936ae0e/1 haElectricalMeasurement.write({"acCurrentDivisor":1000}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

rodionos, 2025-04-15:
The following fix works:

```typescript
        if ("_TZ3000_xkap8wtb" == device.manufacturerName && 162 == device.applicationVersion) {
            await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff"]);
        } else {
            await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "haElectricalMeasurement", "seMetering"]);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
        }   
```

Here's the external converter code that I used: https://gist.github.com/rodionos/c273543e26058712ac68a41ca73a5a0f

I'm not sure why `saveClusterAttributeKeyValue` works even though these attributes are not supported by the device itself.

Koenkk, 2025-04-16:
`saveClusterAttributeKeyValue` bypasses the read from the device (which causes the unsupported attribute error) and directly sets the values in the z2m database. 

Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

rodionos, 2025-04-18:
Referencing [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9202) in the converters repo.

Koenkk, 2025-04-19:
Merged, thanks!

closed 2025-04-19 by Koenkk