Koenkk/zigbee2mqtt #27095: Cannot Find Adapter

ALPF5555, 2025-04-13:
### What happened?

I have upgraded my Zigbee2MQTT and now after the upgrade it cannot find the Herdsman and when I look at the fault it says Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.  I am using the same configuration file that was in the /opt/zigbee2mqtt/data/configuration.yaml 
Is there another file that I should be looking at or configuring.... Here is the error message below 

 pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pnpm start

> zigbee2mqtt@2.2.1 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-13 03:10:35] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-13 03:10:35] info: 	z2m: Starting Zigbee2MQTT version 2.2.1 (commit #a2debc08)
[2025-04-13 03:10:35] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-13 03:10:35] error: 	z2m: Error while starting zigbee-herdsman
[2025-04-13 03:10:35] error: 	z2m: Failed to start zigbee-herdsman
[2025-04-13 03:10:35] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-04-13 03:10:35] error: 	z2m: Exiting...
[2025-04-13 03:10:35] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTcpAdapter (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:522:15)
    at discoverAdapter (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:548:26)
    at Function.create (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/adapter.ts:53:54)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:47)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:104:13)
    at start (/opt/zigbee2mqtt/index.js:149:5)
â€‰ELIFECYCLEâ€‰ Command failed with exit code 1.
pi@raspberrypi:/opt/zigbee2mqtt $

### What did you expect to happen?

I expected that the Zigbee2MQTT would find the adapter as it was working before the upgrade on the same TCP connection and port.

The Configuration.Yaml file is in the same place as it used to be in the /opt/zigbee2mqtt/data/configuration.yaml

The file looks exactly the same as the backup I created before the upgrade.

The Only difference in the files is at the bottom after -184 and before device options there is nothing in the old file there was.

homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

Now in the new configuration.Yaml file it is

device_options: {}

### How to reproduce it (minimal and precise)

Every time I try to start the Zigbee2MQTT it fails with exactly the same message cannot find the Herdsman 

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

2.8.2.dev 3

### Adapter

SLZB-06 

### Setup

I am using a Raspberry Pi 4 with 4GB ram and have been using the same device for over 1 year and the same Adapter working with Zigbee 1.42 for the same time.  The Adapter is POE and I can connect to it with my laptop over the ethernet at the IP address without a problem.

When I try to attach my Configuration. Yaml file it will not allow me it says this type of file is not allowed so I have cut and pasted the first part of the file for you to see
homeassistant:
  enabled: false
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.10:1883
serial:
  port: tcp://192.168.1.147:6638
advanced:
  network_key:
    - 94
    - 175
    - 17
    - 140
    - 240
    - 113
    - 152
    - 119
    - 161
    - 0
    - 230
    - 5
    - 226
    - 223
    - 65
    - 108
  pan_id: 31234
  ext_pan_id:
    - 232
    - 130
    - 22
    - 199
    - 207
    - 81
    - 103
    - 184
device_options: {}

Everything is how it was before the upgrade maybe the Zigbee2MQTT is looking in the wrong file or location for the Adapter?  Has the file location changed or is there something I have missed during the upgrade....

### Debug log

Here is the debug file 

[2025-04-12T15_27_23_190Z-debug-0.log](https://github.com/user-attachments/files/19722680/2025-04-12T15_27_23_190Z-debug-0.log)

ALPF5555, 2025-04-13:
I have found what I cannot find and that is the Onboarding page. It will not allow me go to that Onboarding set up page and it is not located at http://localhost:8080 or any other IP address that I have tried.  I have tried to wipe out the configuration file in the hope that it want to build a new one but this did not work either. I have tried with the frontend in the configuration file enabled true and false that did not make any difference. I cannot open the Onboarding page to be able to set up my adapter and if I try to use my old config file that does not work either.  How can I open the Onboarding Page to setup my adapter etc.

Nerivec, 2025-04-13:
You are missing the `serial` > `adapter` setting. [2.0.0 breaking changes](https://github.com/Koenkk/zigbee2mqtt/discussions/24198#discussion-7273807)
```yaml
serial:
  port: tcp://192.168.1.147:6638
  adapter: zstack
```

al-lac, 2025-04-30:
@Nerivec I have a similar issue, when doing the on-boarding with docker-compose. If I restart the deployment before finishing the setup, it fails to start and does not go to the on-boarding page again until i remove the config that it created automatically:
```yaml
version: 4
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
serial: {}
advanced:
  log_level: info
  channel: 11
  network_key: []
  pan_id: 53346
  ext_pan_id:[]
frontend:
  enabled: true
  port: 8080
homeassistant:
  enabled: false
```

docker-compose.yml:
```yaml
services:
  app:
    image: koenkk/zigbee2mqtt:2.2.1
    restart: on-failure
    volumes:
      - /zigbee2mqtt/data:/app/data:rw
    user: '1000:1000'
    environment:
      ZIGBEE2MQTT_CONFIG_FRONTEND_ENABLED: "true"
      ZIGBEE2MQTT_CONFIG_FRONTEND_PORT: 8080
      Z2M_ONBOARD_NO_REDIRECT: 1
```

Should the on-boarding page not be shown again in case of issues on the initial config?

ALPF5555, 2025-04-30:
Hello Sir             I was never able to get to the onboarding page. I just set the config and made the changes in the config.yaml file. In the server I typed in the Raspberry Pi server I guess in your case it would be the docker container server address.Then in the serial I typed in the Zigbee dongle address mine is a slzb-06 dongle Network attached. 
  port: tcp://192.168.1.147:6638      Your port will be    /dev/ttyUSB0

  adapter: zstack      
  baudrate: 115200 
If you have USB adapter then your serial would be     /dev/ttyUSB0   do not forget to put in the adapter line after the /dev/ttyUsB0.  it will either be zstack or ember once again depending on the type of dongle you are using.
Then the speed buadrate and save the config file and all should work.
Regards Alan Turnbull

closed 2025-04-30 by Nerivec