Koenkk/zigbee2mqtt #27096: Can't reset Energy TS011F_plug_3

Bjk8kds, 2025-04-13:
### What happened?

I can't reset Energy on TS011F_plug_3 device even after following the steps in Dev Console tab, even after I erased and re-paired the device.

Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_okaz9tjs
Description
Smart plug (with power monitoring by polling)
Support status
Supported

Firmware build date
20210625
Firmware version
500+TZSKT31BS102
Manufacturer
Tuya
Model
TS011F_plug_3

[image]

### What did you expect to happen?

Energy reset to 0

### How to reproduce it (minimal and precise)

open Dev Console tab, writes: Endpoint: 1 Cluster: 0x00 Command: 0 Payload: (don't change)

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20250321

### Adapter

zStack

### Setup

Add-on, HA OS x86 baremetal, HP T620

mrmaximas, 2025-04-13:
@Bjk8kds `Dev Console > Endpoint 1 > Cluster 0x00 > Command 0 `then wait some time for the nex report.

Bjk8kds, 2025-04-13:
> [@Bjk8kds](https://github.com/Bjk8kds) `Dev Console > Endpoint 1 > Cluster 0x00 > Command 0 `then wait some time for the next report.

Yeah, tried that bunch of time but nothing happen.

mrmaximas, 2025-04-13:
@Bjk8kds if you load the plug the meter goes up?

Bjk8kds, 2025-04-13:
> @Bjk8kds if you load the plug the meter goes up?

Yes. Power goes up, Energy continues counting.

mrmaximas, 2025-04-13:
@Bjk8kds it's a Tuya ©️🤦🏻‍♂️

la-fouine, 2025-04-13:
Same problem for me, impossible to reset the energy meter to 0 on my 4 TS011F_plug_3 with this method.

Zigbee2MQTT version
2.2.1.1

Adapter
Sonoff ZBDongle-E firmware 7.4.4.0

geoghegan, 2025-06-09:
Same issue for me. Resetting via the MQTT Publish workaround does not work either.

Z2M version 
2.4.0

Adapter
SLZB-06M v2.8.8 / 20250220