Koenkk/zigbee2mqtt #27098: OTA causing ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT

Ricc68, 2025-04-13:
### What happened?

When attempting an OTA, Zigbee2MQTT issues an ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT and bails out.
This is reproducible on 90% of the OTA runs on my setup.
On normal operation z2m is rock solid: only OTA triggers this issue.
Not sure this has been already reported but I've found no mention about this in the open issues.

### What did you expect to happen?

Of course: no timeout issue during OTA.

### How to reproduce it (minimal and precise)

Happens when running an OTA, very frequently like 90% of the times.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

8.0.2

### Adapter

ZBDongle-E

### Setup

Addon installed on HAOS running in Proxmox VM.

### Debug log

[log.log.gz](https://github.com/user-attachments/files/19726385/log.log.gz)

Nerivec, 2025-04-14:
Are you running a recent version of 8.0.2? https://github.com/Nerivec/silabs-firmware-builder/releases (update5 or newer)
The master thread about this was [here](https://github.com/Koenkk/zigbee2mqtt/issues/23861).

Ricc68, 2025-04-14:
Right, I think I'm on older firmware.
Will update it to latest and re-check.
Thanks for the hint!

dan73800, 2025-04-17:
> Are you running a recent version of 8.0.2? https://github.com/Nerivec/silabs-firmware-builder/releases (update5 or newer) The master thread about this was [here](https://github.com/Koenkk/zigbee2mqtt/issues/23861).

Hi
With releases 2.0.0 and higher, is it recommanded to use 8.0.2 firmware on a Sonoff ZBDongle-E rather than 7.4.x ?
I have difficulties to find the list of latest firmwares for my dongle. Have you any URL to purpose ?
Thanks

Nerivec, 2025-04-17:
8.0.2 firmware (latest versions) have fixes for OTA-related troubles (really, high stress scenario). _This is independent of the Z2M version._
You can find the list of firmware I use for [ember-zli](https://github.com/Nerivec/ember-zli/wiki) here:
https://github.com/Nerivec/ember-zli/blob/main/firmware-links.json
_For the most part, it is kept up to date automatically using Github releases from various builders._
Note: ember-zli uses these links [when you are in the bootloader](https://github.com/Nerivec/ember-zli/wiki/Bootloader#update-firmware) and you pick a firmware from the pre-defined list.

closed 2025-04-17 by Nerivec

deezid, 2025-04-21:
> Are you running a recent version of 8.0.2? https://github.com/Nerivec/silabs-firmware-builder/releases (update5 or newer) The master thread about this was [here](https://github.com/Koenkk/zigbee2mqtt/issues/23861).

Can I just update to this using this page?
https://darkxst.github.io/silabs-firmware-builder/

Ricc68, 2025-04-21:
> Since it's at the very end, it might just be that the device is taking an extra-extra long time to finish (OTA end has a very long timeout, for that reason)? Check with `debug` logs if you see any OTA activity still after frozen.

I suspect this was caused by depleted batteries therefore I'll just close the issue.
I will also explore some HA option for sending notifications on batteries below, say, 10%.

I also want to explore some implementation in z2m to allow firmware updates on battery powered devices only if batteries are , say, > 50% because as of now there is no warning that the batteries are low if you run an OTA update.

Nerivec, 2025-04-21:
> I also want to explore some implementation in z2m to allow firmware updates on battery powered devices only if batteries are , say, > 50% because as of now there is no warning that the batteries are low if you run an OTA update.

It would be good, but reported battery levels are not very reliable.
_I have a device that's been at 8% for about a year now..._ 🤣
Some (proper) devices actually check for battery level internally and refuse OTA if it's too low.

Ricc68, 2025-04-28:
@Nerivec I've got a proposal about battery low. We cannot force an update behaviour on battery reported status, it's clear, but what if we just display the power source icon on the left (or the right) of the update button? This would automatically inform the user on the status of the power source.

I'd use the same function implemented in the info page which shows the status with significant icon based on power status.

The rationale is that it is a much better decision if it is an informed decision: would I had pressed the update button if there was a red blinking battery icon? Probably not, unless I know very well my devices like your case of the device at 8%.

Would you support this idea?

Nerivec, 2025-04-28:
Probably would be good to have _something_ about it on the OTA page, just have to figure out what's the best way to inform the user. Maybe add an extra line in the confirm modal if the battery is say below 50%.

Ricc68, 2025-05-02:
@Nerivec I have implemented the icons as per how I imagined them.
This is the implementation rationale:

- <50% is a warning, yellow static icon
- <10% is a danger, red blinking icon
- If there's no % information, no icon (can't do much about it unless we work on PowerSource() but my intent was to be as less intrusive as possible)
- hoovering with the mouse over the icons will give a clear explanation
- we can choose different positioning and size of the icons
- requires very limited code changes (only additions)

[image]

I have looked at the dialog code, but adding the warning in the modal requires modifications to the Button and the DialogConfirmationModal() which is used widely in the code (maybe I have overlooked something here). Alternatively I should implement an OTADialogConfirmationModal() for the Button to use when OTAprompt is set which would only be used in the ota-page.
Neither option is very attractive to me.
Please give your comments on the icons implementation and maybe you have also good suggestions for a clean modal implementation.
I have not committed the code yet for review because I wanted your opinion first, also the other commit is still pending and I want to keep the two things as separate PR.

Nerivec, 2025-05-02:
Probably would go global and put the red icon as soon as battery is `<50%`, `low` or `battery_low` is true (different reported states depending on devices). And not show any icon otherwise.

Ricc68, 2025-05-04:
@Nerivec I created a new PowerSourceOTA implementation in a separate module with minimal impact on index.tsx to show the icons.
This is the result (icons not blinking but is easy to make them blink if you prefer):

[image]

If the graphics is ok I will commit this implementation for your review.