Koenkk/zigbee2mqtt #27102: Unknown values (as specified by ZCL) are being converted to valid but nonsensical values

oddlama, 2025-04-13:
### What happened?

According to the Zigbee Cluster Library (ZCL) specification [[pdf]](https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf), many attributes define reserved values (such as `0x8000` for temperature measurement) that explicitly indicate "unknown" readings:

[image]

Some real world sensors (such as the SEN66 by sensirion) can have extended time periods where some internal measured values may become unavailable (due to calibration periods) where it would be important to report this unknown value instead of a garbage value. The SEN66 specifically has an initial calibration period to accumulate particulate matter measurements, which takes a lot of time (up to 1 hour). In that time it provides other valid data such as temperature and humidity measurements which should be available through zigbee, while the PM2.5 measurement should report that its value is still unknown.

When an attribute is read via z2m that reports the magic unknown value for that attribute, z2m interprets the value as a valid value, even though it lies outside of the range of valid values as specified in ZCL. An unavailable temperature sensor for example will report a value of `-367.68°C`, which is not even physical. Calibration periods can also occur while the sensor is running normally, so it is important to be able to that convey that information to home assistant or any other home automation controller.

### What did you expect to happen?

I'd expect z2m to correctly correctly handle values that are not in the defined range of valid values as given by ZCL, or correspond to a specified special value. Specifically for unknown values I would expect this information to be conveyed via mqtt so the home automation controller knows that a specific sensor is currently unavailable.

A possible workaround would be to prevent client devices from reporting such values in the first place, but unfortunately many downstream zigbee sdks use a precompiled zboss stack (like espressif's or nordic's sdk) which doesn't give developers enough control over the stack to prevent sending values for extended periods of time. So having the unknown value is important to indicate to home automation controllers that the value is nonsensical and should instead be used to mark the relevant sensor as unavailable or ignore the value.

### How to reproduce it (minimal and precise)

Send `0x8000` as a value in a temperature cluster, or any of the unknown values for any other cluster of choice. z2m will blindly interpret the value, even though it is not in the range of valid values as specified by ZCL.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

not applicable

### Adapter

not applicable

### Setup

not applicable

### Debug log

not applicable

Nerivec, 2025-04-15:
I started adding support for non-values a while back, but then 2.0.0 got in the way 😉
I've got most of the rest of it "ready-to-go" (waiting for the move to biome to be over since it will undoubtedly conflict a lot).
We do need to investigate the impact on converters in more details though. and stuff like HA, so these non-values (`Number.NaN` or `undefined`, depending on target) are properly handled/displayed (and don't cause errors...).

oddlama, 2025-04-15:
Sounds great! Let me know if there is a draft ready that can be tested in the field

Nerivec, 2025-08-07:
@oddlama if you get a chance to test this. I couldn't myself, I implemented from spec.