Koenkk/zigbee2mqtt #27106: Tuya Rain Sensor RB-SRAIN01 randomly stopped working

AyraHikari, 2025-04-14:
### What happened?

Availability is online, but the sensor is frozen and no longer update. If i did interviewed it, it does update but only once. After that all sensor will completely frozen again, and i have to reset and rejoin it again to make it working again. But then again it will frozen after few hours later.
[image]

### What did you expect to happen?

Should be working fine like before

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

20240710

### Adapter

CC2652P

### Setup

Add-on on Home Assistant OS on HP Prodesk 600 G2 with KVM

lnlyssg, 2025-04-14:
I think this is a common issue with this sensor rather than anything to do with Zigbee2MQTT, there are multiple reports of it losing connection/going offline in the comments at https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/#comments

jocamane, 2025-04-15:
I have one just like it and so far it has always worked well.

nmaster2042, 2025-04-20:
I have the same issue with mine, and same adapter.

jocamane, 2025-04-20:
Try connect the device direct to the coordinator

nmaster2042, 2025-04-21:
I tried connect the device to the coordinator, and to a router, but at the end, it ends with a disconnexion from the zigbee network.

It's probably not a z2m issue.

NicoLeOca, 2025-04-27:
I experienced the same issues.
Moving from a Combee II to a more recent coordinator fixed it (SMLIGHT SLZB-07 in my case).