Koenkk/zigbee2mqtt #27110: [Feature request]: Improve entity friendly names for TRVZB

onurece, 2025-04-14:
### Is your feature request related to a problem? Please describe

When a new TRVZB device is added, many entities are generated. Two of the entity names are like
1. number.<device_name>_external_temperature_input
2. number.<device_name>_local_temperature_calibration

However, their friendly names are identical
1. \<Device Name\> Temperature
2. \<Device Name\> Temperature

[image]

This is making it difficult to tell which is which and show/hide as needed in history.

See below how both "Sonoff Bedroom TRV Temperature" have different colours in the tooltip and the chart, but the same colour in the legend.
[image]

### Describe the solution you'd like

Their friendly names should be different. A solution would be like below so that their friendly names would be closer to their entity names.
1. "\<Device Name\> External Temperature Input "
or this could match how it's written in z2m and be "\<Device Name\> External Temperature"
[image]

2. "\<Device Name\> Local Temperature Calibration"

### Describe alternatives you've considered

I can try to rename all entities, but it wouldn't be a solution, but a workaround, and all the users would have to repeat the same thing every time they add a new device.

### Additional context

I looked for where this name is generated in the code, but couldn't find it. So, I hope this has something to do with z2m and not Sonoff.

onurece, 2025-04-14:
23.20.1 which I believe is the latest one.

The friendly name of the calibration entity has never had any information about it being for calibration, but it wasn’t a huge pain. So, I’ve never complained. Now with the addition of an entity with same name it is confusing.

mrmaximas, 2025-04-14:
@lcheng33775823 сan you please fix the converter?

onurece, 2025-07-16:
I've force removed and repaired, but it's still the same.

Koenkk, 2025-07-18:
This is because in case the device class is set (temperature in this case), the entity name is generated automatically. Changing this behaviour would break other stuff so I suggest for this case to manually rename the sensors.