Koenkk/zigbee2mqtt #27115: [Wrong device]:

alexebner, 2025-04-14:
### Link

https://www.zigbee2mqtt.io/devices/50208702.html#sunricher-50208702

### Model

50208702

### Description

Is recognized wrong as     SR-ZG9030A-MW

### Vendor

Sunreacher

### Picture (link)

-

### Database entry

-

### Notes

Some of my LED driver are recognized as wrong device. The devices in the picture are the same devices.

[image]

mrmaximas, 2025-04-14:
please find and post the database.db entry for this device.
database.db is usually located in the /zigbee2mqtt/data folder

alexebner, 2025-04-14:
Wrong entry:
`{"id":42,"type":"Router","ieeeAddr":"0x9035eafffe65db62","nwkAddr":18328,"manufId":4107,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"ZG9030A-MW","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,65535,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ZG9030A-MW","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r63"}},"genOta":{"attributes":{"currentFileVersion":63}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254,"onLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r63","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1744653757591}`

Right entry:
{"id":55,"type":"Router","ieeeAddr":"0x9035eafffe65db6c","nwkAddr":4939,"manufId":4107,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"HK-DIM","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,65535,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"HK-DIM","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r63"}},"genOta":{"attributes":{"currentFileVersion":63}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254,"onLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r63","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1744653666404}

chris-1243, 2025-04-14:
According to the device definition in herdsman-converter, you should be able to modify the modelID due to an error made by Sunricher in some batch.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/96efe558ebe43ba52a70a2a049d157a27cad15df/src/devices/sunricher.ts#L762

https://github.com/Koenkk/zigbee-herdsman-converters/blob/96efe558ebe43ba52a70a2a049d157a27cad15df/src/devices/sunricher.ts#L1120-L1129

alexebner, 2025-04-14:
Thank you for the info.
That has worked

closed 2025-04-14 by alexebner