Koenkk/zigbee2mqtt #27122: suite of issue  #26937

steflp, 2025-04-14:
### What happened?

Hello  
I  try to update from 2.1.3 to  2.2.1 
I  update node js  to  22.14 ( #nodejs -v)
 but  when  I start my zigbee alone (  with this command  #pnpm start my system work  fine .

When  I  want to start it with  systemctl   
I ve these error  

`avril 14 22:07:14 meteo-Ubuntu pnpm[143133]: > zigbee2mqtt@2.2.1 start /opt/zigbee2mqtt
avril 14 22:07:15 meteo-Ubuntu pnpm[143150]: TypeError: Module "file:///opt/zigbee2mqtt/package.json" needs an import assertion of type "json"
avril 14 22:07:15 meteo-Ubuntu systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
avril 14 22:07:15 meteo-Ubuntu systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.`

 I  give you  my  systemctl file : 

`[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
ExecStart=/usr/local/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
systemd.exec(5)
StandardError=inherit
Restart=always
RestartSec=10s

[Install]
WantedBy=multi-user.target`

I roll back my config to  2.1.3 and its wordk fine 

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20250321

### Adapter

SonOff  Zlb 06 Zlb06M

### Setup

Works on Ubuntu 22.04

Nerivec, 2025-04-15:
Any chance you have two `node` versions installed and your systemctl is triggering with the wrong one?
- `node --version`
- `/usr/local/bin/node --version` (or somewhere else)

Note: you can replace the cmd to be more direct (once you know the proper `node` path):
`ExecStart=node index.js`

steflp, 2025-04-15:
# nvm ls-remote
# nvm install 22.14.0

root@meteo-Ubuntu:~# node -v
v22.14.0

nstefy, 2025-04-18:
Did u try to use in systemctl file

`ExecStart=/usr/bin/node index.js`

instead of

` ExecStart=/usr/local/bin/npm start`

steflp, 2025-04-18:
Hello 
Sorry  no time at the moment to test this ;-( sorry 
Possible to  suspend ?
Thanks

steflp, 2025-04-18:
OK  ,   I  take few minutes to  upgrade from  2.1.3 to  2.2.1 and now is OK 
 but  with nodejs (???) 
 
`root@meteo-Ubuntu:/opt/zigbee2mqtt-LAN2TEST# nodejs -v
v22.14.0
root@meteo-Ubuntu:/opt/zigbee2mqtt-LAN2TEST# which nodejs
/usr/bin/nodejs
root@meteo-Ubuntu:/opt/zigbee2mqtt-LAN2TEST# cat  /etc/systemd/system/zigbee2mqttLAN2.service 
[Unit]
Description=zigbee2mqttLAN2
After=network.target

[Service]
Environment=NODE_ENV=production
#ExecStart=/usr/local/bin/npm start
ExecStart=/usr/bin/nodejs index.js
WorkingDirectory=/opt/zigbee2mqtt-LAN2TEST
StandardOutput=inherit
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling syslog, for more options see systemd.exec(5)
StandardError=inherit
Restart=always
RestartSec=10s
#User=steflp

[Install]
WantedBy=multi-user.target
root@meteo-Ubuntu:/opt/zigbee2mqtt-LAN2TEST#`

closed 2025-04-18 by steflp

steflp, 2025-04-19:
all my  zigbee is uptodate to  2.2.1 with nodejs 22.14