Koenkk/zigbee2mqtt #27124: [Wrong device]: AVATTO ZWSM16-2-Zigbee recognized as Tuya TS0002

Gera1974, 2025-04-15:
### Link

https://avatto-smart.com/product/smart-switch-module-zwsm16/

### Model

 ZWSM16-2-Zigbee

### Description

2 gang switch module

### Vendor

 AVATTO

### Picture (link)

https://www.zigbee2mqtt.io/devices/ZWSM16-2-Zigbee.html#avatto-zwsm16-2-zigbee

### Database entry

{"id":96,"type":"Router","ieeeAddr":"0xa4c1384fcfaff925","nwkAddr":18271,"manufId":4417,"manufName":"_TZ>

mrmaximas, 2025-04-17:
pleaser post full db entry

Gera1974, 2025-04-23:
See below the full database.db file as requested !
Thanks
===================================================================================

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c2b665","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1744507737853}
{"id":7,"type":"EndDevice","ieeeAddr":"0x000d6f00168a35e9","nwkAddr":41642,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"c700000202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":360}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":90}},"closuresDoorLock":{"attributes":{"lockState":1}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":900,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":11,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1745447519861,"checkinInterval":90}
{"id":8,"type":"Router","ieeeAddr":"0x847127fffe29b722","nwkAddr":12037,"manufId":4098,"manufName":"_TZ3000_8kzqqzu4","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,10,4,5,6,258],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"tuyaBacklightMode":0}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":100,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":0,"moesCalibrationTime":133}},"genBasic":{"attributes":{"65503":"Az /\u0011","65506":27,"modelId":"TS130F","manufacturerName":"_TZ3000_8kzqqzu4","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1745447551701}
{"id":9,"type":"Router","ieeeAddr":"0x5c0272fffe0a3bd1","nwkAddr":43397,"manufId":4098,"manufName":"_TZ3000_rdtixbnu","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"seMetering":{"attributes":{"currentSummDelivered":239930,"divisor":100,"multiplier":1}},"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_rdtixbnu","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":127,"rmsCurrent":19,"activePower":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAd [... cut ...] f":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1745447329277}

mrmaximas, 2025-04-24:
DB Entry is
{"id":96,"type":"Router","ieeeAddr":"0xa4c1384fcfaff925","nwkAddr":18271,"manufId":4417,"manufName":"_TZ3000_mtnpt6ws","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"manuSpecificTuya_4":{"attributes":{"inching":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}},"genBasic":{"attributes":{"65503":"\u001c\u0000\u0000\u0000\u0012"\b /\u0012","65506":56,"65508":0,"65534":0,"manufacturerName":"_TZ3000_mtnpt6ws","zclVersion":3,"appVersion":72,"modelId":"TS0002","powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b665","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":260,"epId":242,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1745447513392}

closed 2025-07-01 by github-actions[bot]