Koenkk/zigbee2mqtt #27125: 🚨 After updating Zigbee2MQTT [version] sensors stopped working

Ovchinnikov77, 2025-04-15:
### What happened?

After updating from 1.39.1 to the latest version of Zigbee2MQTT (2.2.1), sensors started to fail from time to time, namely, disappear, not show status and go offline (usually a reboot helped). I had to roll back to 1.39.1.

### What did you expect to happen?

I expected the new version to be more stable.
I hope the bug will be fixed.

### How to reproduce it (minimal and precise)

With the old version [1.39.1], everything worked fine, but after updating to 2.2.1, when using Ember or EZSP, failures began to occur, namely, ROXIMO Water Leak sensors fell off from time to time and routers [Tuya Air Box Smart ZigBee 5 in 1 Air Quality Detector] worked unstably.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

7.4.5.0 build 0

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Zigbee2MQTT add-on in Home Assistant
Running on Raspberry Pi 4
(Home Assistant OS – output of uname not available)

### Debug log

Unfortunately I can't provide the log.

chinswain, 2025-04-15:
Did you try adding a device? I have a very similar issue but cannot roll back (HAOS Add-on). I ended up reinstalling HAOS and zigbee2mqtt which did not resolve the severe instability.

Gertjuh, 2025-04-15:
I also have problems with most of the sensors, but I don't think it is a Z2M problem. I can see in the Zigbee2MQTT screen the PIR's are reacting. So I suspect Home Assistant to have an issue. I updated HA yesterday.

ATM-HSW, 2025-04-16:
I use Z2M in a VM, without any other services. I upgraded from 1.x to 2.x in March and have been having problems ever since. In the beginning everything went relatively well, but problems started to accumulate. I have just rebuilt everything, new Z2M installation, new Zigbee network set up because the entire network has become unusable.

The network structure looks good, in the first moment, the connections between coordinator and routers are established. The sensors are connected to the routers or the coordinator. But a few hours later, the routers are offline again. Restarting the routers works partially, dedicated routers in most cases, bulbs need to be reset to reconnect to the network. Some sensors continue to work stably despite this.

The network is not big, it had 70+ nodes, currently I have it down to 40+. I have been using SLZB-06 for a long time, but have also tried SLZB-06M. Same problems with both. Node is the current version, SLZB-06(M) has been using the current stable version and I tried dev versions. Only Zigbee stack is used, no matter.

I checked and changed the Zigbee channels. Same behavior/problems. And the channel scans done by SLZB-06 look OK.

I will try to revert to the last version 1.x.

lnlyssg, 2025-04-18:
Did you follow all the breaking change guidance when upgrading? https://github.com/Koenkk/zigbee2mqtt/discussions/24198

ATM-HSW, 2025-04-18:
My last step was a complete new installation: new VM (Debian), new Z2M, resetting/pairing all devices from scratch.
It seems that only a few devices are going offline (IKEA light bulb LED1623G12, Lidl light HG08008, IKEA power socket E160x/E170x/E190x). Both ikea devices just need a power off/on, to be usable again. The lidle light needs a new pairing. I didn't know the problems with IKEA Parasoll E2013, but they are removed from the network since a few hours. The problem with going offline of the above devices is still existing. I checked the logs - there are no error messages between the last sucessful communication and errors because of the offline state.
I wasn't sure whether the changes in the configuration from the breaking change guidance were also necessary for a new installation, but adding them didn't change anything.

closed 2025-06-25 by github-actions[bot]