Koenkk/zigbee2mqtt #27126: Tuya Power Meter signature issue

jaromanda, 2025-04-15:
### What happened?

I tried pairing a Tuya power meter https://www.aliexpress.com/item/1005005466228073.html which came back as "NOT supported"

On looking through the codebase, I see the following

    fingerprint: tuya.fingerprint("TS0601", ["_TZE204_ac0fhfiq"])

Which looks remarkably like my device, and from my investigations that should be correct, however my device has

    fingerprint: tuya.fingerprint("TS0601", ["_TZE284_ac0fhfiq"])

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

8.0.2[GA]

### Adapter

SMLight SLZB-06Mg24

### Setup

Add-on Home Assistant x86-64 KVM

### Debug log

```
info 2025-04-15 11:14:06z2m: Zigbee: allowing new devices to join.
info 2025-04-15 11:14:06zh:ember: [STACK STATUS] Network opened.
info 2025-04-15 11:14:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"ik5da-1"}'
info 2025-04-15 11:14:18zh:controller: Interview for '0xa4c1389efc728432' started
info 2025-04-15 11:14:18z2m: Device '0xa4c1389efc728432' joined
info 2025-04-15 11:14:18z2m: Starting interview of '0xa4c1389efc728432'
info 2025-04-15 11:14:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389efc728432","ieee_address":"0xa4c1389efc728432"},"type":"device_joined"}'
info 2025-04-15 11:14:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389efc728432","ieee_address":"0xa4c1389efc728432","status":"started"},"type":"device_interview"}'
info 2025-04-15 11:14:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389efc728432","ieee_address":"0xa4c1389efc728432"},"type":"device_announce"}'
info 2025-04-15 11:14:20zh:controller: Succesfully interviewed '0xa4c1389efc728432'
info 2025-04-15 11:14:20z2m: Successfully interviewed '0xa4c1389efc728432', device has successfully been paired
warning 2025-04-15 11:14:20z2m: Device '0xa4c1389efc728432' with Zigbee model 'TS0601' and manufacturer name '_TZE284_ac0fhfiq' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2025-04-15 11:14:20z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1389efc728432/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1389efc728432"],"manufacturer":"_TZE284_ac0fhfiq","model":"Automatically generated definition","model_id":"TS0601","name":"0xa4c1389efc728432","via_device":"zigbee2mqtt_bridge_0x90395efffecdc60b"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c1389efc728432_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.2.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1389efc728432","unique_id":"0xa4c1389efc728432_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2025-04-15 11:14:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601","options":[],"supports_ota":false,"vendor":"_TZE284_ac0fhfiq"},"friendly_name":"0xa4c1389efc728432","ieee_address":"0xa4c1389efc728432","status":"successful","supported":false},"type":"device_interview"}'
```

Koenkk, 2025-04-15:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-04-15 by Koenkk

jaromanda, 2025-04-16:
I can confirm that the commit has solved the issue.

Thank you for the speedy resolution.