Koenkk/zigbee2mqtt #27131: ZWT198 Wall Thermostat: running_state not being picked up by MQTT Auto Discovery

Clumpton, 2025-04-16:
### What happened?

This thermostat exposes a "running_state" with values of "idle" and "heat" corresponding to whether the thermostat has switched the heating (e.g. boiler) off or on. This item is not picked up by Domoticz MQTT Auto Discovery despite appearing in Z2M as below:
`{ "backlight_mode": "high", "child_lock": "UNLOCK", "current_heating_setpoint": 22.5, "deadzone_temperature": 0.5, "factory_reset": "OFF", "frost_protection": "ON", "linkquality": 90, "local_temperature": 22.5, "local_temperature_calibration": 0.1, "max_temperature_limit": 29.5, "preset": "auto", "running_state": "heat", "schedule_holiday": "07:00/22.0°C 23:00/18.5°C", "schedule_weekday": "06:00/21.5°C 08:00/22.5°C 11:30/21.0°C 12:30/22.0°C 17:00/22.0°C 22:00/18.5°C", "sensor": "internal", "system_mode": "heat", "working_day": "7" }`

For info
`export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_xnbkhhdr',`

MQTT Explorer shows this for "climate"

TOPIC
`homeassistant/climate/0xa4c138948f5233eb/climate/config`

VALUE
`{
  "action_template": "{% set values = {None:None,'idle':'idle','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ values[value_json.running_state] }}",
  "action_topic": "zigbee2mqtt/New Lounge Thermostat",
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "current_temperature_template": "{{ value_json.local_temperature }}",
  "current_temperature_topic": "zigbee2mqtt/New Lounge Thermostat",
  "device": {
    "hw_version": 1,
    "identifiers": [
      "zigbee2mqtt_0xa4c138948f5233eb"
    ],
    "manufacturer": "Tuya",
    "model": "Wall thermostat",
    "model_id": "ZWT198/ZWT100-BH",
    "name": "New Lounge Thermostat",
    "via_device": "zigbee2mqtt_bridge_0x00124b001cde54dc"
  },
  "max_temp": "35",
  "min_temp": "5",
  "mode_command_topic": "zigbee2mqtt/New Lounge Thermostat/set/system_mode",
  "mode_state_template": "{{ value_json.system_mode }}",
  "mode_state_topic": "zigbee2mqtt/New Lounge Thermostat",
  "modes": [
    "off",
    "heat"
  ],
  "name": null,
  "object_id": "new_lounge_thermostat",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "2.2.1-dev",
    "url": "https://www.zigbee2mqtt.io"
  },
  "preset_mode_command_topic": "zigbee2mqtt/New Lounge Thermostat/set/preset",
  "preset_mode_state_topic": "zigbee2mqtt/New Lounge Thermostat",
  "preset_mode_value_template": "{{ value_json.preset }}",
  "preset_modes": [
    "manual",
    "auto",
    "temporary_manual"
  ],
  "temp_step": 0.5,
  "temperature_command_topic": "zigbee2mqtt/New Lounge Thermostat/set/current_heating_setpoint",
  "temperature_state_template": "{{ value_json.current_heating_setpoint }}",
  "temperature_state_topic": "zigbee2mqtt/New Lounge Thermostat",
  "temperature_unit": "C",
  "unique_id": "0xa4c138948f5233eb_climate_zigbee2mqtt"
}`

### What did you expect to happen?

Expected state "running_state" to appear in Domoticz devices.

### How to reproduce it (minimal and precise)

Add the thermostat to Zigbee2MQTT,
See all the exposed states & settings appear in Domoticz. Note that running_state is absent.

### Zigbee2MQTT version

2.2.1-dev commit: [b694dc15]

### Adapter firmware version

 20210708

### Adapter

Slaesh's CC2652RB Stick

### Setup

Z2M installed as legacy on Debian Bullseye
Linux DomoticzMain 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
Raspbian GNU/Linux 11

Raspberry Pi4 running Domoticz, Z2M, nodered for a couple of Tasmota devices

### Debug log

[log.log](https://github.com/user-attachments/files/19774984/log.log)

Koenkk, 2025-04-16:
Given that the HA discovery payload is correctly published, this is likely a bug in Domoticz, I would recommend submitting an issue there.

Clumpton, 2025-04-16:
> Given that the HA discovery payload is correctly published, this is likely a bug in Domoticz, I would recommend submitting an issue there.

I was sent here from there!
If you look at my post you can indeed see running_state in the payload, however it does not appear in the Values found by MQTT explorer.
Incidentally is it normal for the topic (homeassistant/climate/0xa4c138948f5233eb/climate/config) to have "climate" twice?

Koenkk, 2025-04-17:
The `running_state` is part of the climate discovery:

```
"action_template": "{% set values = {None:None,'idle':'idle','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ values[value_json.running_state] }}", "action_topic": "zigbee2mqtt/New Lounge Thermostat"
```

But it seems Domoticz ignores that?

Clumpton, 2025-04-17:
> The `running_state` is part of the climate discovery:
>
> ```
> "action_template": "{% set values = {None:None,'idle':'idle','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ [...]

Looks like that is the case. Where do I go from here?

Koenkk, 2025-04-18:
Domoticz needs to add support for this.

Clumpton, 2025-04-18:
OK, bounced it back to Domoticz. Thanks