Koenkk/zigbee2mqtt #27133: [New device support]: Zemismart ZMS-206US-3

avi9899, 2025-04-16:
### Link

https://www.zemismart.com/products/zms-206us?srsltid=AfmBOoq-6cXhedS2o1PaeT2hyQTiUEKQUIvCrnqvGKePf5LwWSmCjftP

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xa4c138c434afd8a7","nwkAddr":29665,"manufId":4417,"manufName":"_TZE204_iyki9kjp","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1744817170000}

### Zigbee2MQTT version

2.2.1

### Comments

hello, 
there is a similar device that already supported in latest update.
the only defernite is the shape (EU vs US) of the device. 
https://www.zigbee2mqtt.io/devices/ZMS-206EU-3.html#zemismart-zms-206eu-3

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_iyki9kjp',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I failed to use the external definition at all...

Koenkk, 2025-04-17:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE204_iyki9kjp` to `_TZE204_k7v0eqke`
- Start Z2M and see if the device now works

avi9899, 2025-04-20:
It's works! thank you very much!

closed 2025-04-20 by avi9899

Koenkk, 2025-04-21:
Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

You can change it back after updating to the next release.

mpekin1, 2025-04-28:
Any chance of getting this for the Zemismart ZMS-206US-4 (4 Gang Switch)

Koenkk, 2025-07-04:
@thedeadliestcatch added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.