Koenkk/zigbee2mqtt #27136: service is restarting after a couple of seconds. pnpm start runs fine

SteffenH123, 2025-04-16:
### What happened?

I installed on rasp 3 on Ubuntu 24.04.2 LTS along Linux service description.
When starting on console it runs fine and stable. 
When starting as service, service will be restarted around every 30 seconds.

https://www.zigbee2mqtt.io/guide/installation/01_linux.html

### How to reproduce it (minimal and precise)

install service on Rasp 3 with ubuntu LTS and setup as service.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

7.4.3

### Adapter

EmberZNet

### Setup

plain install on rasp 3 and Ubuntu 24.04.2 LTS 

https://www.zigbee2mqtt.io/guide/installation/01_linux.html

Nerivec, 2025-04-17:
Can you check the `debug` logs?

francisp2, 2025-06-17:
??

closed 2025-08-25 by github-actions[bot]