Koenkk/zigbee2mqtt #27140: [Problem Report] Smartthings Multipurpose Sensor IM6001-MPP02 Wrong Naming & Sensors Not Reporting

K-BIB, 2025-04-17:
### What happened?

First thing noticed, device is physically labelled as IM6001-MPP02 but IM6001-MPP01 in Zigbee2MQTT device list. https://www.zigbee2mqtt.io/devices/IM6001-MPP01.html

Main issue is echoing mostly; https://github.com/Koenkk/zigbee2mqtt/issues/25277

Pairing using instructions; "When pairing, make sure to keep the sensor awake for 20 seconds by opening and closing the contact every second. Reset Procedure - https://www.youtube.com/watch?v=yT3iVHuO7Qk" Several resets completed

### These are not reporting anything:
battery (Null %)
moving (Null)
X axis (Null)
Y axis (Null)
Z axis (Null)

### These are working:
temperature
contact
battery low
tamper
linkquality

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

TBC

### Adapter

SLZB-06M

### Setup

Mini PC plain

toxic0berliner, 2025-05-03:
I just realized #25277 is closed so I can report I have the exact same behavior.
If someone with skills is able to find a solution that would be awesome 👍 
For what it's worth someone seems to get them working after pairing with zha. Maybe something in the pairing/interview process isn't done right here.

thepiecesofme, 2025-05-25:
I am also having the exact same issues with this sensor. I posted on this at the previous issue link here: https://github.com/Koenkk/zigbee2mqtt/issues/25277 but as others have reported, it’s since been closed so no further help. I need to purchase some additional multi-sensors like this but am concerned since these don’t seem to fully work at this point. Oddly, the older versions of this same sensor (different shape but were sold under the same SmartThings Multi Sensor name) all work perfectly.

Any help would be hugely appreciated. Thanks in advance!

toxic0berliner, 2025-08-03:
Oups, forgot about this, I guess this is in the release now, will try that.

toxic0berliner, 2025-10-05:
Sorry for the delay, can only sadly report that it still doesn't work for me :(

kenobob, 2026-01-15:
@Koenkk i too am having the same issue. Apparently that fix didn't work. I'm mid converting from smartthings to z2m, hit my first multi sensor and the same, no motion or battery. (this sensor worked fine in smartthings just an hour ago.)

[image]

toxic0berliner, 2026-01-15:
I recently re-paired it with z2m 2.7.2 and confirm a similar behavior. I do get battery though but no XYZ position. Never got it either with z2m or zha and an don't own a smartthings hub to verify the sensor but I got it new so no reason it shouldn't work. 

[image]

84zwqpd56q-tech, 2026-02-13:
Now temperature and battery report on the sensor is fine, after re pairing a few times, but still none of the sensor have vibration and axis working

logankp, 2026-02-16:
I did some digging, looks like zigbee-herdsman v8 and above have this issue. I was able to use 7.0.6 and it worked just fine. I tried to `git bisect` my way through but I'm not certain the commit I landed on was actually the culprit.

logankp, 2026-02-17:
I pulled the zigbee2mqtt repo, the zigbee-herdsman repo, and used `pnpm link` to use the herdsman repo as the package. Check out the v7.0.6 tag and build. I did that so I could attempt to track down the commit that caused the issue, but you could probably just change the `zigbee-herdsman` package version in `packages.config`.

logankp, 2026-02-19:
I sniffed the traffic between z2m and the device when attempting to configure reporting for the accelerometer data through the z2m UI. I figured out what's causing the `UNSUPPORTED_ATTRIBUTE` error: when configuring through the UI the manufacturerCode isn't set properly and ends up being sent as `0x110a` [1]. Sending the configure_reporting request to the topic manually [2]though with the correct manufacturer ID of `0x1241` [3] succeeds and I'm able to get accelerometer data afterwards.

[1]. [image]

[2]. `{"id":"<device_id>","endpoint":"1","cluster":"manuSpecificSamsungAccelerometer","attribute":"x_axis","minimum_report_interval":1,"maximum_report_interval":900,"options":{"manufacturerCode": "0x1241"}}`

[3]. [image]

There's still a bug here because the device isn't configured properly upon initial join, but after configuring reporting manually the accelerometer data shows.

Koenkk, 2026-02-21:
Fixed by @Nerivec !

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2026-02-21 by Koenkk