Koenkk/zigbee2mqtt #27148: Samotech 314 zigbee dimmer - now unsupported?

keithskillicorn, 2025-04-18:
### What happened?

I have 2 samotech 314 zigbee dimmer modules that have been working perfectly for 3 years. 2 weeks ago they both disappeared from network. I have tried to rejoin, but Z2M reports them as unsupported.

### What did you expect to happen?

Continue to function, able to rejoin network.

### How to reproduce it (minimal and precise)

Can't reproduce them disappearing. I can consistently reproduce them not joining my network.

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20240710

### Adapter

Sonoff dongle plus P

### Setup

Home assistant add-on

Koenkk, 2025-04-19:
Could you provide a screenshot of the z2m frontend -> device > about page?

keithskillicorn, 2025-04-20:
Was eventually able to get them to rejoin the network.

closed 2025-04-20 by keithskillicorn