Koenkk/zigbee2mqtt #27151: Table_full error

sdetweil, 2025-04-18:
### What happened?

I have two devices that are temp/humidity sensors one living room, one bathroom, (turns on fan when humidity rises)
from HA, Bathroom sensor
```text
Xiaomi temperature & humidity sensor with custom firmware (LYWSD03MMC)
by Custom devices (DiY)
Connected via [Zigbee2MQTT Bridge](https://ha.detweiler.xyz/config/devices/device/089decb27d5a401ba520bcab14cf6c1a)
Firmware: 3001-0123
Hardware: 20
```

they connect, but one (this one) produces an error , from the log
```
[2025-04-18 07:20:40] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=198 Len=27]
[2025-04-18 07:20:40] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler: type=UNICAST apsFrame={"profileId":0,"clusterId":32801,"sourceEndpoint":0,"destinationEndpoint":0,"options":320,"groupId":0,"sequence":223} packetInfo:{"senderShortId":44118,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":136,"lastHopRssi":-66,"lastHopTimestamp":0} messageContents=018c
[2025-04-18 07:20:40] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=TABLE_FULL, payload=undefined
[2025-04-18 07:20:40] debug: 	zh:controller:endpoint: Error: Bind 0xa4c1380a35acccbd/1 msTemperatureMeasurement from '0xe0798dfffebbe238/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.5.2/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:574:23)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at runNextTicks (node:internal/process/task_queues:69:3)
    at processImmediate (node:internal/timers:459:9)
    at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.29.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:40:9)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.29.0/node_modules/zigbee-herdsman-converters/src/devices/custom_devices_diy.ts:871:13)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.29.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)
[2025-04-18 07:20:40] error: 	z2m: Failed to configure 'master bathroom sensor', attempt 1 (Error: Bind 0xa4c1380a35acccbd/1 msTemperatureMeasurement from '0xe0798dfffebbe238/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.5.2/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:574:23)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at runNextTicks (node:internal/process/task_queues:69:3)
    at processImmediate (node:internal/timers:459:9)
    at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.29.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:40:9)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.29.0/node_modules/zigbee-herdsman-converters/src/devices/custom_devices_diy.ts:871:13)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.29.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

and both devices show in HA correctly

### What did you expect to happen?

no error

### How to reproduce it (minimal and precise)

hard to do without the sensor

### Zigbee2MQTT version

latest0dev (4-18-25)

### Adapter firmware version

Firmware: 2.2.1-dev Hardware: EmberZNet 7.4.4 [GA]

### Adapter

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231007145920

### Setup

docker, on amd 64
Linux buildserver 5.4.0-212-generic #232-Ubuntu SMP Sat Mar 15 15:34:35 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 20.04.6 LTS

### Debug log

[~zigbee.log](https://github.com/user-attachments/files/19812695/zigbee.log)

sdetweil, 2025-04-18:
[image]

Nerivec, 2025-04-19:
Usually means the device did not self-clean properly. You might be able to fix that by doing a factory reset (procedure can be different by model, but a longer hold on the pairing button might do the job, double the time it takes to trigger pairing and hopefully that works...).
_Sometimes you may not see the impact of a failed binding if you don't really use the resulting effect, but sometimes it will break some features._

sdetweil, 2025-04-19:
but it should have a throw stack(says unhandled) , really a message would be fine..

Nerivec, 2025-04-19:
After a full factory reset, it should not be giving this status (table should be cleared). Either the device is misbehaving, or it wasn't fully reset.

closed 2025-06-26 by github-actions[bot]