Koenkk/zigbee2mqtt #27152: TS0044 can pair but no switch key on device

qin-nz, 2025-04-18:
### What happened?

I have read https://github.com/Koenkk/zigbee2mqtt/issues/15749 , I can pair it.

There are only battary is device panel  and @MacArtiom and @quadrupole1988 also mention at the end of https://github.com/Koenkk/zigbee2mqtt/issues/15749

[image]

### What did you expect to happen?

It should be 4 button or switch on device panel.

### How to reproduce it (minimal and precise)

Just pair TS0044

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

add on on HA OS on x86

Koenkk, 2025-04-19:
Could you provide the data/database.db entry of this device?

tungmeister, 2025-04-21:
@Koenkk I'm having the same issue with this device after taking the plunge and upgrading to 2.x. 
Everything is showing correctly in zigbee2mqtt:

[image]

and I can see the button state being published:
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Buttons/action', payload '4_double'

but there is no action within hass:

[image]

afterburnerx0, 2025-04-25:
> [@Koenkk](https://github.com/Koenkk) I'm having the same issue with this device after taking the plunge and upgrading to 2.x. Everything is showing correctly in zigbee2mqtt:
> 
> [image]
> 
> and I can see the button state being published:
> 
> `z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Buttons/action', payload '4_double'`
> 
> but there is no action within hass:
> 
> [image]
> 
> EDIT: ignore this, my config file was messed up and the legacy action sensor setting wasn't being read. All working now.

I am facing the same issue, could you please share where to fix the CONFIG file please? I cannot see the action sensor in MQTT. Thank you!!! =)

tungmeister, 2025-04-25:
@afterburnerx0 see the home assistant section [here](https://github.com/Koenkk/zigbee2mqtt/discussions/24198) - it's the legacy_action_sensor.

afterburnerx0, 2025-04-26:
amazing mate, thank you! ill get onto this.

closed 2025-07-03 by github-actions[bot]