Koenkk/zigbee2mqtt #27155: [New device support]: Zigbee 24V Low Voltage Thermostat TH1420ZB for Hydronic Radiant Floor

fraserb2, 2025-04-18:
### Link

https://www.amazon.com/dp/B0CYQ9GQL9?ref=ppx_yo2ov_dt_b_fed_asin_title

### Database entry

{"id":46,"type":"Router","ieeeAddr":"0x500b91400006140a","nwkAddr":58919,"manufId":4508,"manufName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":"TH1420ZB-01\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513,516,1026,1794,2821,65281],"outClusterList":[10,65281,25],"clusters":{"genBasic":{"attributes":{"modelId":"TH1420ZB-01\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":17,"stackVersion":8,"hwVersion":0,"dateCode":"20231117\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"1381\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2353,"tolerance":50}},"seMetering":{"attributes":{"currentSummDelivered":0,"divisor":1000,"multiplier":1}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe930e76","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe930e76","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":17,"stackVersion":8,"hwVersion":0,"dateCode":"20231117\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"1381\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1745002204169}

### Zigbee2MQTT version

2.2.1

### Comments

Only exposes temp, power, and energy. not able to utilize the thermostat function. It is a rebranded sinope TH1400ZB which i also have that works great 

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TH1420ZB-01                    '],
    model: 'TH1420ZB-01                    ',
    vendor: 'Sinope Technologies            ',
    description: 'Automatically generated definition',
    extend: [m.temperature(), m.electricityMeter({"cluster":"metering"})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I was not able to figure out how to use the external definition

fraserb2, 2025-04-21:
I was able to get the external definition to finally load. All I can see is temperature, power, and energy. None of the actual functions of the thermostat are functional. I have tried loading the sinope ts file in as the convertor but it does not work. If there is something else I can try to do to troubleshoot please let me know. my computer background is limited though.

lit-af, 2025-04-28:
This is a rebranded Sinopé thermostat. Since other thermostats of the same product line are supported, it shouldn't be hard to make it work with z2m. https://www.zigbee2mqtt.io/devices/TH1400ZB.html

fraserb2, 2025-04-29:
> This is a rebranded Sinopé thermostat. Since other thermostats of the same product line are supported, it shouldn't be hard to make it work with z2m. https://www.zigbee2mqtt.io/devices/TH1400ZB.html

I do actually have one of these as well and they work well. What should I do to make this one work?

lit-af, 2025-05-10:
I wish I could help, I think it might just be a model number issue, but I might be wrong.

In the mean time, I found this list that could help: https://github.com/claudegel/sinope-zha