Koenkk/zigbee2mqtt #27157: zh:zstack:znp: Socket error Error: read ECONNRESET

bapesupreme, 2025-04-19:
### What happened?

[2025-04-18 21:46:54] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-18 21:46:54] info:     z2m: Starting Zigbee2MQTT version 2.2.1 (commit #unknown)
[2025-04-18 21:46:54] info:     z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-18 21:46:54] info:     zh:zstack:znp: Opening TCP socket with 192.168.1.205:6638
[2025-04-18 21:46:54] info:     zh:zstack:znp: Socket connected
[2025-04-18 21:46:54] info:     zh:zstack:znp: Socket ready
[2025-04-18 21:46:54] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-04-18 21:46:55] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-04-18 21:46:56] error:    zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-04-18 21:46:56] info:     zh:zstack:znp: Port closed
[2025-04-18 21:47:01] error:    z2m: Error while starting zigbee-herdsman
[2025-04-18 21:47:01] error:    z2m: Failed to start zigbee-herdsman
[2025-04-18 21:47:01] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-04-18 21:47:01] error:    z2m: Exiting...
[2025-04-18 21:47:01] error:    z2m: Error: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:104:13)
    at start (/opt/zigbee2mqtt/index.js:149:5)

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20221226

### Adapter

SLZB-06 - tried v2.8.5.dev7 and v2.7.0 and 2.8.2dev3

### Setup

Proxmox

### Debug log

[image]

bapesupreme, 2025-04-19:
downgrading to v2.6.8.dev26

mattmoo, 2025-04-21:
I'm having the same issue with the same setup (except Docker, not Proxmox). Downgrading did not solve it, zigbee2mqtt crashes after a couple of minutes.

bartje, 2025-04-21:
Same issue as @bapesupreme (proxmox --> VM --> docker) but his sollution didn't work

`
2025-04-21T13:26:21.625499345Z [2025-04-21 13:26:21] error: 	zh:zstack:znp: Socket error Error: read ETIMEDOUT
2025-04-21T13:26:21.626543722Z [2025-04-21 13:26:21] error: 	z2m: Adapter disconnected, stopping
2025-04-21T13:26:22.039655416Z Using '/app/data' as data directory
2025-04-21T13:26:22.085091398Z Starting Zigbee2MQTT without watchdog.
2025-04-21T13:37:22.104691063Z [2025-04-21 13:37:22] error: 	zh:zstack:znp: Socket error Error: read ETIMEDOUT
2025-04-21T13:37:22.105633744Z [2025-04-21 13:37:22] error: 	z2m: Adapter disconnected, stopping
2025-04-21T13:37:22.566759984Z Using '/app/data' as data directory
2025-04-21T13:37:22.614446927Z Starting Zigbee2MQTT without watchdog.
`

bapesupreme, 2025-04-21:
Other possible solutions

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/644#issue-2576150187

MysticalMount, 2025-06-14:
I didnt get this that often when running as root, since switching to run as rootless, seems to consistently give the issue, still digging though so please take with a pinch of salt.  Might be some kind of capability required at some point.

closed 2025-08-22 by github-actions[bot]