Koenkk/zigbee2mqtt #27158: Take rid of spammy devices

danirebollo, 2025-04-19:
### What happened?

Hi, I'm using MQTT since some years with no problems, but now I've added some new devices:

https://www.zigbee2mqtt.io/devices/TS0601_smart_air_house_keeper.html#tuya-ts0601_smart_air_house_keeper
https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html#tuya-zy-m100-s_2 (ceiling type)

There devices report 4 times a second, and reports linkquality as the only change.
{
    "detection_delay": 2.4,
    "fading_time": 2,
    "last_seen": "2025-04-19T07:32:30.510Z",
    "linkquality": 171,
    "maximum_range": 3.9,
    "minimum_range": 0.6,
    "presence": false,
    "radar_sensitivity": 4,
    "target_distance": 0,
    "illuminance": null
}

I have filtered this attributes on configuration.yaml, but the messages remain (now linkquality not changes, nothing changes, but it is updated and last_seen changes every 200ms
```
filtered_attributes:
      - linkquality
```

How to deal with this devices?
I'm waiting to change this devices for WiFi parts and control with ESPHome so behave like I want, but the real problem is that my Zigbee instance (tried on both docker and LXC) grows in RAM and CPU until crash. Maybe because the high amount of messages trying to get analized, and also my zigbee network works worst. 
Z2M filters works only after message is received, so the problem in the network congestion remains the same. 
Whould be great, at least, two things:
- some kind of alert or log can be sent to the user warning that spammy device is detected and Z network whould be works better if removed. 
- Fail safe behaviour. Can't crash. At least reboot. I have a script thar reboots when RAM goes high, this is bruteforce but at leats keeps my Z2M running most of the time:

```
#!/bin/sh

THRESHOLD=85

get_mem_stats() {
  top -b -n1 | awk '/^Mem:/ {
    used=$2
    free=$4
    total=used + free
    percent = int((used * 100) / total)
    printf "%d %d %d\n", used, free, percent
  }'
}

read used free percent <<EOF
$(get_mem_stats)
EOF

echo "[Z2M Watchdog] Memoria usada: ${used}K | libre: ${free}K | uso: ${percent}%"

if [ "$percent" -ge "$THRESHOLD" ]; then
  echo "[Z2M Watchdog] RAM al ${percent}%, esperando 10s para confirmar..."
  sleep 10

  read used2 free2 percent2 <<EOF
  $(get_mem_stats)
EOF

  echo "[Z2M Watchdog] (Confirmación) usada: ${used2}K | libre: ${free2}K | uso: ${percent2}%"

  if [ "$percent2" -ge "$THRESHOLD" ]; then
    echo "[Z2M Watchdog] RAM sigue alta ($percent2%). Reiniciando Zigbee2MQTT..."
    rc-service zigbee2mqtt restart
  else
    echo "[Z2M Watchdog] Falsa alarma. RAM ahora al ${percent2}%"
  fi
else
  echo "[Z2M Watchdog] RAM OK al ${percent}%"
fi
```

### What did you expect to happen?

Avoid crashes

### How to reproduce it (minimal and precise)

Add spammy devices like:
https://www.zigbee2mqtt.io/devices/TS0601_smart_air_house_keeper.html#tuya-ts0601_smart_air_house_keeper
https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html#tuya-zy-m100-s_2 (ceiling type)
And see hoy RAM and CPU increases over time (in 2 hours you can see the sawtooth wave on RAM and CPU )

### Zigbee2MQTT version

2.2.1 commit: a14feef8c13f5ce49b0ac358d98d3234d21d7ff4 also tested in 1.4

### Adapter firmware version

20240710

### Adapter

sonoff zigbee 3.0 usb dongle plus zStack3x0

### Setup

x86 miniPC -> USB conected to it -> proxmox -> LXC -> Z2M

Tee86, 2025-04-20:
try...
```
retain: false
optimistic: true
debounce: 3
filtered_attributes:
   - illuminance
   - linkquality
```

Tee86, 2025-04-20:
@danirebollo what mqtt broker are you using? Mosquitto?

> ...but Zigbee network remains saturated..

that is not exactly true. Yep, those devices keep on yapping twice per sec, but no one listens to them, so it doesn't really congest the network. I can even prove it, as this week I realized that 4 Moess thermostats are spamming one of my Zigbee networks. If you look at messages in/out from my broker, you can see very clearly exactly when I applied the very exact fix I sent you above...

[image]

CPU temp went down by 15℃ since then...

danirebollo, 2025-04-20:
@Tee86 I'm using Mosquitto, and works well, it can handle all the messages.
Your screenshots are from your broker and shows that limiting messages on software side, mqtt receives in fact less messages, which it's ok because less cpu, hassio works better etc...

But you need zigbee sniffer to analize a zigbee network. I haven't sniffed mine but the reality is that MAC access errors grow since this spamming devices are connected and also pairing takes more attempts to do (before only need one). 
If two devices are sending messages each 250ms ( 8 messages each second ) the gap to let other devices send his messages is smaller. In fact maybe even all the other devices power consumption increases due to reattempts sending data or waiting for empty slot.

closed 2025-06-27 by github-actions[bot]