Koenkk/zigbee2mqtt #27160: Read response failed

pierrepinon, 2025-04-19:
### What happened?

I have severeal Tuya devices like [TH02Z](https://www.zigbee2mqtt.io/devices/TH02Z.html#tuya-th02z) who spawn error in log with following messages : 

zh:controller:device: Read response to 0xa4c138d46c281c2b failed (ZCL command 0xa4c138d46c281c2b/1 genTime.readRsp({"localTime":798385687}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":232,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))

We have many routers in my network all room is covered by one routeur.

### What did you expect to happen?

No error in log and no communication problem

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20230507

### Adapter

TI CC2652P Coordinator

### Setup

Add-on home assistant

### Debug log

`
[2025-04-19 08:47:37] error: 	z2m: Failed to read state of 'Ampoule Chambre' after reconnect (ZCL command 0x287681fffe1e95c5/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 09:38:11] error: 	zh:controller:device: Read response to 0xa4c1380ac09c03df failed (ZCL command 0xa4c1380ac09c03df/1 genTime.readRsp({"localTime":798370676}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":151,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 09:38:26] error: 	zh:controller:device: Read response to 0xa4c1380ac09c03df failed (ZCL command 0xa4c1380ac09c03df/1 genTime.readRsp({"localTime":798370677}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":151,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 09:38:41] error: 	zh:controller:device: Read response to 0xa4c1380ac09c03df failed (ZCL command 0xa4c1380ac09c03df/1 genTime.readRsp({"localTime":798370678}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":151,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 09:48:12] error: 	zh:controller:device: Read response to 0xa4c138d46c281c2b failed (ZCL command 0xa4c138d46c281c2b/1 genTime.readRsp({"localTime":798371278}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":214,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 09:48:28] error: 	zh:controller:device: Read response to 0xa4c138d46c281c2b failed (ZCL command 0xa4c138d46c281c2b/1 genTime.readRsp({"localTime":798371279}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":214,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 09:48:42] error: 	zh:controller:device: Read response to 0xa4c138d46c281c2b failed (ZCL command 0xa4c138d46c281c2b/1 genTime.readRsp({"localTime":798371280}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":214,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 11:43:46] error: 	zh:controller:device: Read response to 0xa4c13889d282a1db failed (ZCL command 0xa4c13889d282a1db/1 genTime.readRsp({"localTime":798378211}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":203,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 11:44:01] error: 	zh:controller:device: Read response to 0xa4c13889d282a1db failed (ZCL command 0xa4c13889d282a1db/1 genTime.readRsp({"localTime":798378212}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":203,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 11:44:15] error: 	zh:controller:device: Read response to 0xa4c13889d282a1db failed (ZCL command 0xa4c13889d282a1db/1 genTime.readRsp({"localTime":798378213}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":203,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 12:39:16] error: 	zh:controller:device: Read response to 0xa4c138d5dc0fe93c failed (ZCL command 0xa4c138d5dc0fe93c/1 genTime.readRsp({"localTime":798381541}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":232,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 12:39:30] error: 	zh:controller:device: Read response to 0xa4c138d5dc0fe93c failed (ZCL command 0xa4c138d5dc0fe93c/1 genTime.readRsp({"localTime":798381542}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":232,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 12:39:45] error: 	zh:controller:device: Read response to 0xa4c138d5dc0fe93c failed (ZCL command 0xa4c138d5dc0fe93c/1 genTime.readRsp({"localTime":798381543}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":232,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 13:39:57] error: 	zh:controller:device: Read response to 0xa4c1380ac09c03df failed (ZCL command 0xa4c1380ac09c03df/1 genTime.readRsp({"localTime":798385182}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":169,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 13:40:12] error: 	zh:controller:device: Read response to 0xa4c1380ac09c03df failed (ZCL command 0xa4c1380ac09c03df/1 genTime.readRsp({"localTime":798385183}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":169,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 13:40:27] error: 	zh:controller:device: Read response to 0xa4c1380ac09c03df failed (ZCL command 0xa4c1380ac09c03df/1 genTime.readRsp({"localTime":798385184}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":169,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 13:48:21] error: 	zh:controller:device: Read response to 0xa4c138d46c281c2b failed (ZCL command 0xa4c138d46c281c2b/1 genTime.readRsp({"localTime":798385686}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":232,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-04-19 13:48:36] error: 	zh:controller:device: Read response to 0xa4c138d46c281c2b failed (ZCL command 0xa4c138d46c281c2b/1 genTime.readRsp({"localTime":798385687}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":232,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
`

pierrepinon, 2025-04-24:
I found solution. I rebind device with closer router and no error is triggered

closed 2025-04-24 by pierrepinon

christianwicke, 2025-06-22:
I have the same problem. I also have a CC2652P2 coordinator (firmware probably 20230507) and some routers (using the same usb dongles with router firmware 20221102). And I also have Tuya devices ([WSD500A](https://www.zigbee2mqtt.io/devices/WSD500A.html#tuya-wsd500a) and [TS0203](https://www.zigbee2mqtt.io/devices/TS0203.html#tuya-ts0203)).

Some of the devices produce the exact same error since a couple month. E.g.:
```
[2025-06-21 08:39:06] error:    zh:controller:device: Read response to 0xa4c138ed830ab88d failed (ZCL command 0xa4c138ed830ab88d/1 genTime.readRsp({"localTime":803810298}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":255,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
[2025-06-21 08:39:31] error:    zh:controller:device: Read response to 0xa4c138ed830ab88d failed (ZCL command 0xa4c138ed830ab88d/1 genTime.readRsp({"localTime":803810299}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":255,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
```
But these errors come and go and it is sometimes the one device and then the other.

I didn't bind any device explicitly to a router yet. I thought zigbee is handling all that for me and finds the closest router automatically. Is it really necessary to do explicit binds? Or is this some kind of bug?