Koenkk/zigbee2mqtt #27161: [New device support]: Support for Legrand NLJ - Garage door

jordiinstalaciones, 2025-04-19:
### Link

https://www.legrand.es/es/productos/micromodulo-conectado-para-puertas-garaje-with-netatmo-067696?category_id=159301

### Database entry

{"id":65,"type":"Router","ieeeAddr":"0x0004740000277395","nwkAddr":41612,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" NLJ - Garage door\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,258,4,5,15,64513],"outClusterList":[0,64513,5,258,25],"clusters":{"genBasic":{"attributes":{"modelId":" NLJ - Garage door\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":67,"hwVersion":1,"dateCode":" \u000020231207\u0000\u0000\u0000\u0000\u0000","swBuildId":"0007"}},"genBinaryInput":{"attributes":{}},"closuresWindowCovering":{"attributes":{"61446":0,"61447":2,"currentPositionLiftPercentage":100,"tuyaMovingState":0}}},"binds":[{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b0031de9bbc","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":67,"hwVersion":1,"dateCode":" \u000020231207\u0000\u0000\u0000\u0000\u0000","swBuildId":"0007","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1745074037184}

### Zigbee2MQTT version

2.2.1 commit: unknown

### Comments

Hi! I'm trying to support a Zigbee device in my network. I followed the instructions on the "supporting new devices" page, but I got stuck and I’m not sure how to proceed.

The device is a Legrand NLJ – Garage door. I already got its interview data from Zigbee2MQTT and generated the external definition automatically.

I’ve added the definition below. Could you help me understand what I need to fix or improve to make it work properly?

Thanks in advance!

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: [' NLJ - Garage door             '],
    model: ' NLJ - Garage door             ',
    vendor: ' Legrand                       ',
    description: 'Automatically generated definition',
    extend: [m.windowCovering({"controls":["lift","tilt"]}), m.binary({"name":"binary_input_1","cluster":"genBinaryInput","attribute":"presentValue","reporting":{"attribute":"presentValue","min":"MIN","max":"MAX","change":1},"valueOn":["ON",1],"valueOff":["OFF",0],"description":"binary_input_1","access":"STATE_GET","endpointName":"1"}), m.commandsWindowCovering()],
    meta: {},
};
```

### What does/doesn't work with the external definition?

The device shows up in Zigbee2MQTT and appears to be correctly identified. However, the binary input doesn’t seem to update in real time, and the control via “lift” or “tilt” doesn’t actually open or close the garage door.

I’m not sure if I need a custom converter or just to adjust the current definition.

Thank you!

jordiinstalaciones, 2025-04-19:
Here is the manual for the device in case it helps:

[LE13977AB.pdf](https://github.com/user-attachments/files/19821695/LE13977AB.pdf)

jordiinstalaciones, 2025-05-01:
Hi @Koenkk, would you mind taking a look at this device support request when you have a moment? I followed the steps, but I’m not sure if a custom converter is needed. Thanks a lot for your help!

Koenkk, 2025-05-03:
>  and the control via “lift” or “tilt” doesn’t actually open or close the garage door.

Does setting the position also not work? If not, you will need to sniff traffic with the original gateway to reverse engineer this: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

jcorrod17, 2025-05-09:
Also interested in this device. Any help?

jordiinstalaciones, 2025-05-10:
Hi! Thanks again for the support.

Just to clarify everything regarding this garage door device:

The lift command works — it correctly opens and closes the door.
The tilt command doesn’t apply, since this is a garage door and cannot tilt.
For position control, dragging the position slider triggers the door to open from around 50% and up, but there is no accurate control or real position feedback — it just starts opening past the 50% mark.

About the 
binary sensors
 (magnetic contacts):

The door has two separate binary sensors, not just one.
One detects when the door is fully closed, and the other when it’s fully open.
When the door starts opening, as soon as it leaves the “closed” contact, it should enter a transitional state, like “opening” or “moving”.
When it reaches the “open” contact, it should then be marked as “open”.
When closing, the same logic applies in reverse: leaving the “open” contact should change to “closing”, and reaching the “closed” contact should mark it as “closed”.

Current issue:

This logic is unreliable — sometimes the “open” state is not detected, and this causes another big issue:

When you press the “open” button, and the door is already open (but not detected as such), nothing happens.
Same when you press “close” — if the system doesn’t correctly know it’s open, it doesn’t act.
This happens frequently, and the root cause seems to be that the system doesn’t truly know the current state of the door.

About the Zigbee2MQTT integration:

In the current Zigbee2MQTT interface, only one binary input appears (binary_input_1), and it’s shown as N/A, as you can see in this screenshot:
It seems like the second binary input is missing or not recognized by the converter. However, the hardware definitely has two physical contacts, one for “open” and another for “closed”. Both should be exposed and handled properly to ensure reliable state detection and automation.

Thanks again for your support — any help would be very appreciated!

Jordi Jordá MonchoGerenteJORDI INSTALACIONES S.C.Instalaciones de climatización, electricidad, energía solar y domóticaRúa pardo bazan 46 · Burela (Lugo)Tel.: 982 582 ***@***.***: www.jordiinstalaciones.es