Koenkk/zigbee2mqtt #27167: [New device support]: TS0726 2-gang light switch and 4-gang light switch

MReuter2203, 2025-04-20:
### Link

https://de.aliexpress.com/item/1005008666629595.html?spm=a2g0o.order_list.order_list_main.33.241e5c5fwIIdRq&gatewayAdapt=glo2deu

### Database entry

{"id":76,"type":"Router","ieeeAddr":"0xa4c1386cc51d9bee","nwkAddr":15256,"manufId":4417,"manufName":"_TZ3002_zjuvw9zf","powerSource":"Mains (single phase)","modelId":"TS0726","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","modelId":"TS0726","manufacturerName":"_TZ3002_zjuvw9zf","powerSource":1,"zclVersion":3,"appVersion":85,"hwVersion":1}},"genOnOff":{"attributes":{"65278":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"65278":0,"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x048727fffe1ef931","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"65278":0,"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"65278":0,"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x048727fffe1ef931","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1745139023648}

### Zigbee2MQTT version

2.2.1

### Comments

I installed my two new devices and paired them with Zigbee. They were detected immediately but marked as "not supported".
So I created a converter file using the information from the developer console and by clicking on generated_external_definition.
I then integrated the code into my file and restarted Zigbee2MQTT. The file was recognized, and the "not supported" status disappeared. So far, so good.

Only L1 is connected to my N (neutral) and L (live) cables, so I can switch the light on and off using Switch 1.
However, when I control one of the switches via Home Assistant, both switches react. In other words, both switches either turn on or off at the same time.
Manual operation of the physical buttons works separately and as expected.

Because of this behavior, I am unable to create automations or control the switches properly via Alexa.

What other information do you need to help me solve this problem?

[image]

Thank you.

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

module.exports = {
    zigbeeModel: ['TS0726'],
    model: 'TS0726',
    vendor: '_TZ3002_zjuvw9zf',
    description: 'Automatically generated definition',
    extend: [
        m.deviceEndpoints({"endpoints":{"1":1,"2":2}}),
        m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]})
    ],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

when I control one of the switches via Home Assistant, both switches react. In other words, both switches either turn on or off at the same time.

Koenkk, 2025-04-21:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZ3002_zjuvw9zf` to `_TZ3002_1s0vfmtv`
- Start Z2M and see if the device now works

MReuter2203, 2025-04-22:
@Koenkk , after modifying the database.db, I now see a device picture (although it shows the wrong module) and a few more settings became available, like Switch Mode 1 and Switch Mode 2.
However, the core issue remains unchanged: pressing one switch still triggers both.
I also a 4-gang switch with vendor id, _TZ3002_1s0vfmtv where only 2 switches are shown...

MReuter2203, 2025-04-23:
4-gang is now showing 4 switches, because my converter file for the 2-gang was the same for this. 
@Koenkk more information for you:
Automations works well when I bind it directly to the switch 1 or 2 from the light switch and then press the button directly on the light switch. 
But when I create an dashboard button, using a physical button or use Amazon Alexa, both switches are triggered and also then all automations from the switches.

Koenkk, 2025-04-26:
> 4-gang is now showing 4 switches, because my converter file for the 2-gang was the same for this.

What did you change? Did you modify the converter?

> Automations works well when I bind it directly to the switch 1 or 2 from the light switch and then press the button directly on the light switch.
But when I create an dashboard button, using a physical button or use Amazon Alexa, both switches are triggered and also then all automations from the switches.

If I understand correctly:
- When pushing the buttons on the device itself, it works (1 gang is toggled)
- When controlling via z2m both gangs toggle

MReuter2203, 2025-04-27:
> > 4-gang is now showing 4 switches, because my converter file for the 2-gang was the same for this.
> 
> What did you change? Did you modify the converter?

I had the same converter for both switches. That was the issue from my side

> If I understand correctly:
>
> * When pushing the buttons on the device itself, it works (1 gang is toggled)
> * When controlling via z2m both gangs toggle

Yes! 
I also tried it with a Tuya zigbee gateway. I added the gateway via my smart life app and added the 2-gang switch via the gateway. Then the HACS Tuya integration in home assistant found and added the 2-gang switch but it was shown as unsupported and no entities (switches) where shown, too.

MReuter2203, 2025-04-29:
> > I had the same converter for both switches. That was the issue from my side
> 
> What converter are you now using for the 4 gang?
I am using the same but modified it a little bit via chatgpt, so that I have 4 endpoint.
> 
> > When controlling via z2m both gangs toggle
> 
> I cannot explain why this happens, could be that we need to do some extra magic here. Any chance you could sniff the traffic from pairing it to the z2m gateway until toggling 1 switch?
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html
Not really. 
I can give you some json (payloads) information from tuya (developer portal). Maybe this is helpfull for you?
One strange thing is, when I try to manuelly change the endpoint via zigbee2mqttm, only endpoint 1 is shown in the dropdown list.
It seems that, my switch TS0726 technically has only ONE Zigbee endpoint (Endpoint 1), not four separate ones.
Maybe this means (answer from ChatGPT):
There is only a single actual endpoint (1) in the Zigbee world.
However, the device pretends externally to have 4 switches.
These 4 "virtual" switches are not treated as real endpoints, but are likely distinguished using a shared data point (DP).
👉 That's why your current "multiEndpoint" converter doesn't work.

---
But the solution / update from ChatGPT for the converter did not work.
Hopefully, you have some magic

adikus1201, 2025-04-29:
I have the same problem, I've tried ZHA and zigbee2MQTT and no solution has helped so far :(

zielaq, 2025-04-29:
i'm also struggling with this issue, but for 2-gang i just solved it. adding external converter works for me. now i can switch on and off only one switch (earlier - when `no support` both was switched).
Now also i can detach switch each switch from relay, works great

adikus1201, 2025-04-29:
can you tell me how you did it?

MReuter2203, 2025-04-29:
@zielaq , can you paste your converter, please?
I also have the issue when I load 2 converter file only one is used by the 4 and 2-gang switches.
So that means 4 switches will be displayed by the 2-Gang switch

zielaq, 2025-04-29:
sure, but i'm not sure if that is fully properly done. for me just started working.
Needed steps: 
1. create directory `external_converters` in `zigbee2mqtt` directory
2. in created `external_converters` directory, create new File `ts-0726.js`
3. content of the file: 
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: "TS0726",
            manufacturerName: "_TZ3002_zjuvw9zf",
        }
    ],
    model: 'TS0726',
    vendor: 'Bseed',
    description: 'Bseed - Two Gang Switch',
    powerSource: 'Mains (single phase)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        tz.on_off,
        tuya.tz.datapoints
    ],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.action(['scene_1', 'scene_2']),  // Scene control
        e.linkquality(),
        tuya.exposes.switch().withEndpoint('switch_1'),  // Switch 1
        tuya.exposes.switch().withEndpoint('switch_2'),  // Switch 2
        e.enum('relay_status_1', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 1'),  // Power-on behavior for Switch 1
        e.enum('relay_status_2', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 2'),  // Power-on behavior for Switch 2
        e.enum('mode_1', ea.STATE_SET, ['switch_1', 'scene_1']).withDescription('Mode for Switch 1'),  // Mode for Switch 1
        e.enum('mode_2', ea.STATE_SET, ['switch_2', 'scene_2']).withDescription('Mode for Switch 2'),  // Mode for Switch 2
    ],
    endpoint: (device) => {
        return {
            'switch_1': 1,
            'switch_2': 2
        };
    },
    meta: {
        multiEndpoint: true,
        disableDefaultResponse: true,
        timeout: 5000,
        tuyaDatapoints: [
            [1, 'scene_1', tuya.valueConverter.static('scene')],
            [2, 'scene_2', tuya.valueConverter.static('scene')],
            [17, 'scene_id_group_id', tuya.valueConverter.raw],
            [18, 'mode_1', tuya.valueConverterBasic.lookup({'switch_1': 0, 'scene_1': 1})],
            [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch': tuya.enum(0), 'scene': tuya.enum(1)})],
            [24, 'switch_1', tuya.valueConverter.onOff],
            [25, 'switch_2', tuya.valueConverter.onOff],
            [39, 'relay_status_1', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 1
            [40, 'relay_status_2', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 2
        ],
    },
};

module.exports = definition;`

4. save 
5. restart z2mqtt.
6. remove unsupported devices and attach them again to network

zielaq, 2025-04-30:
sorry guys - i pasted wrong file. here is right one

[ts-0726.txt](https://github.com/user-attachments/files/19972399/ts-0726.txt)

MReuter2203, 2025-04-30:
@zielaq , holy cow.... that´s working for me for the 2-gang switch. You are my lifesaver. My wife gave me time until the weekend and then I would have had the choice between a head shorter or reverting everything back to manual light switches. thank you. Now I'll see if I can do the same for the 4-gang switch

bigroma73, 2025-05-09:
@zielaq Thank so much!! My 3 gang bseed works great now.

my file with little changes if someone needs it

[bseed3.txt](https://github.com/user-attachments/files/20128351/bseed3.txt)

Koenkk, 2025-05-10:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

VBenkovskyy, 2025-05-18:
A few hours with ChatGPT, and it finally WORKS!
4 gang—each button works independently!

Don´t forget: 

> create directory external_converters in zigbee2mqtt directory
> in created external_converters directory, create new File bseed-4Gang.js
> content of the file:
> _bseed-4Gang.txt_
> 
> save
> restart z2mqtt.
> remove unsupported devices and attach them again to network

[bseed-4Gang.txt](https://github.com/user-attachments/files/20270231/bseed-4Gang.txt)

makc9, 2025-05-18:
Thanks. The same trick worked for a 3-|Gang switch _TZ3002_iedhxgyi.

pano360, 2025-05-24:
> [@zielaq](https://github.com/zielaq) Thank so much!! My 3 gang bseed works great now.
> 
> my file with little changes if someone needs it
> 
> [bseed3.txt](https://github.com/user-attachments/files/20128351/bseed3.txt)

I thank you all guys. Special thanks to @bigroma73 for the 3 gang switch definition. I just received the newest (model 2025) BSEED 3-gang click switch and it was unsupported. I didn't even expect to find a solution right today. Now my switch works correctly in HA.

Koenkk, 2025-05-27:
I added the various models, thanks all!

Could somebody make a pr for the pictures?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TS0726_1_gang.png` and `TS0726_3_gang.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

VBenkovskyy, 2025-05-28:
Thank you! Done. 
https://github.com/VBenkovskyy/zigbee2mqtt.io/tree/VBenkovskyy-BSEED-TS0726-1-2-3-4-Gang-Images

VBenkovskyy, 2025-05-28:
> [@VBenkovskyy](https://github.com/VBenkovskyy) could you also make a pull request?

https://github.com/VBenkovskyy/zigbee2mqtt.io/pull/1
Done

VBenkovskyy, 2025-06-05:
> I added the various models, thanks all!
>
> Could somebody make a pr for the pictures?
>
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) [...]

Can you please use my images? Now it is from Relay and not from 1-2-3-4-Gang-Switch... Thanks!

[image]

adikus1201, 2025-07-17:
Is it possible to add a double-click or long-click option to these buttons to trigger different scenes? I can't trigger entities if the button is in scene mode.

david200x, 2025-08-04:
Would it be possible to transform the Quirk code to work with the ZHA system?

nicseng, 2025-12-12:
Hi all, I have attempted to use the .js files uploaded by you. On the latest version of Z2M edge. However, Z2M repeatedly included a .invalid behind the js file, rendering it unused. May I know what has changed and how I could edit the js file to make it work with mine?