Koenkk/zigbee2mqtt #27169: Unable to add devices to empty zigbee group

pfremm, 2025-04-20:
### What happened?

Adding a device to the group times out and does not get added.

### What did you expect to happen?

Device get added to the group.

### How to reproduce it (minimal and precise)

Create a group and attempt to add a device via UI

### Zigbee2MQTT version

 2.2.1-1

### Adapter firmware version

2.7.1

### Adapter

SLZB-06

### Setup

Addon via home assistant

### Debug log

fo 2025-04-20 16:13:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{},"error":"Failed to add from group (ZCL command 0x7cc6b6fffe9bbf82/1 genGroups.add({\"groupid\":55,\"groupname\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Timeout - 27592 - 1 - 2 - 4 - 0 after 10000ms))","status":"error","transaction":"c59tx-1"}'
error 2025-04-20 16:13:49z2m: Failed to add from group (ZCL command 0x7cc6b6fffe9bbf82/1 genGroups.add({"groupid":55,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 27592 - 1 - 2 - 4 - 0 after 10000ms))

pfremm, 2025-04-20:
I am attempting to add smartwings blinds to a z wave group and that seems to be causing the issue.  I can add bulbs to another group I guess just fine.  Are groups limited to specific device types or features?

mundschenk-at, 2025-05-01:
Z-Wave?

pfremm, 2025-05-01:
No it seems to only occur with my window blinds that are zigbee.

mundschenk-at, 2025-05-01:
I was wondering because you wrote "... to a z wave group".

closed 2025-07-09 by github-actions[bot]