Koenkk/zigbee2mqtt #27180: All devices lost after upgrade

jeanbruder, 2025-04-21:
### What happened?

I upgraded 3 things on a "test" environment :
- LIDLBox firmware upgrade to 7.5.0 (using the upgrade package provided by jnilo1) : OK
- docker container upgrade from 1.4.2 to 2.2.1 : OK
- adapter change from "ezsp" to "ember" : OK
After that, everything is working fine : WEB-UI showing the values for the devices connected.

Production environment :
- LIDLBox firmware upgrade to 7.5.0 (using the upgrade package provided by jnilo1) : OK
- docker container upgrade from 1.4.2 to 2.2.1 : OK
- adapter change from "ezsp" to "ember" : OK
After that, I lost all my devices from the WEB-UI.

### What did you expect to happen?

Expected to upgrade the production environment without issues like the test environment.

### How to reproduce it (minimal and precise)

Do the upgrade in production, devices are lost.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

7.5.0 [GA]

### Adapter

LIDLBox

### Setup

Docker container

### Debug log

[zigbee2mqtt_2025-04-22.00-17-41.log](https://github.com/user-attachments/files/19840772/zigbee2mqtt_2025-04-22.00-17-41.log)

jeanbruder, 2025-04-21:
Trying to "reconfigure" is not working with the following error for one of the devices :
[2025-04-22 00:27:15] error: 	z2m: Failed to configure '0x70b3d52b6014660d', attempt 1 (Error: ZCL command 0x70b3d52b6014660d/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '15298'.)
    at EmberAdapter.onMessageSent (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:459:42)
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31)
    at Ezsp.emit (node:events:524:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5043:14)
    at Ezsp.callbackDispatch (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:726:26)
    at Ezsp.onAshFrame (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:386:22)
    at UartAsh.emit (node:events:524:28)
    at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1171:45)
    at processImmediate (node:internal/timers:491:21))
[2025-04-22 00:27:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{},"error":"Failed to configure (ZCL command 0x70b3d52b6014660d/1 genBasic.read([\"manufacturerName\",\"zclVersion\",\"appVersion\",\"modelId\",\"powerSource\",65534], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Delivery failed for '15298'.))","status":"error","transaction":"1pqdb-6"}'
[2025-04-22 00:27:10] info: 	z2m: Configuring '0x70b3d52b6014660d'

Trying to "interview" the same device :
[2025-04-22 00:28:25] info: 	z2m: Interviewing '0x70b3d52b6014660d'
[2025-04-22 00:29:09] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of '0x70b3d52b6014660d' (0x70b3d52b6014660d) failed: Error: Interview failed because can not get active endpoints ('0x70b3d52b6014660d')'
[2025-04-22 00:29:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of '0x70b3d52b6014660d' (0x70b3d52b6014660d) failed: Error: Interview failed because can not get active endpoints ('0x70b3d52b6014660d')","status":"error","transaction":"1pqdb-7"}'

Removing (with force) the device and pairing again works.

chrisrock1984, 2025-04-23:
same issue on my side - devices start to disappear

closed 2025-04-30 by jeanbruder