Koenkk/zigbee2mqtt #27181: Zigbee2Mqtt can't remove device

jwhite, 2025-04-22:
### What happened?

Currently on the home assistant addon version  2.2.1-1

My door sensor quit working and I chose to remove it and add it back to zigbee2mqtt.

Other devices continue to work.

Removing the device has the following error:

```
[2025-04-22 12:07:13] info: 	z2m: Removing device 'Office Door' (block: false, force: false)
[2025-04-22 12:07:18] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'Office Door' (block: false, force: false) (Error: Delivery failed for '21824'.)'
[2025-04-22 12:07:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device 'Office Door' (block: false, force: false) (Error: Delivery failed for '21824'.)","status":"error","transaction":"4eemb-1"}'

I would love to downgrade the version, but not sure how

### What did you expect to happen?

I expected that the device be removed

### How to reproduce it (minimal and precise)

Clicked on the trash can in the zigbee2mqtt gui, got an error that quickly disappeared (delivery failed.)

Log is above.

### Zigbee2MQTT version

Zigbee2MQTT version 2.2.1 commit: unknown

### Adapter firmware version

8.0.2 [GA]

### Adapter

sonoff stick

### Setup

Add-on in home assistant

### Debug log

[2025-04-22 12:07:13] info: 	z2m: Removing device 'Office Door' (block: false, force: false)
[2025-04-22 12:07:18] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'Office Door' (block: false, force: false) (Error: Delivery failed for '21824'.)'
[2025-04-22 12:07:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device 'Office Door' (block: false, force: false) (Error: Delivery failed for '21824'.)","status":"error","transaction":"4eemb-1"}'

chris-1243, 2025-04-22:
This an end device. You must wake it up before sending any commands.

Now, as it is not in your network anymore, you should force removed it then you have to factory reset it

jwhite, 2025-04-22:
Happy to mark this closed.  I didn't realise the 'force' was a slider in my UI as the button surround blends into the background.  I thought it was telling me it was already set for force.

closed 2025-04-22 by jwhite