Koenkk/zigbee2mqtt #27182: [New device support]: GreenPower_2 Tuya powerless switch K9Z(TY)

choovick, 2025-04-22:
### Link

https://vi.aliexpress.com/item/1005005469765581.html

### Database entry

{"id":45,"type":"GreenPower","ieeeAddr":"0x00000000f82c8f58","nwkAddr":36696,"modelId":"GreenPower_2","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1745282255 441,"gpSecurityKey":[25,5,70,22,44,197,27,12,107,88,217,4,81,31,58,85]}

### Zigbee2MQTT version

2.2.1 commit: unknown (via addon)

### Comments

Device not reporting anything, do i need to follow this? https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

not clear in the docs which file these definition live in... UI does not save them.

Device model on the sticker: K9Z(TY)

### External definition

```shell
import {genericGreenPower} from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    fingerprint: [{modelID: 'GreenPower_2', ieeeAddr: new RegExp('^0x00000000f82c8f58$')}],
    model: 'GreenPower_2',
    vendor: '',
    description: 'Automatically generated definition for Green Power',
    extend: [genericGreenPower()],
};
```

### What does/doesn't work with the external definition?

Sorry new to this, I don't think anything works. I'm not sure which file to update and try other definition.

chris-1243, 2025-04-22:
Please check #19405

Support for this device is still complicated for the time

choovick, 2025-04-22:
Ok I'll keep the conversation on that thread, but for others who might want to try this device, this worked for me.

1. Create `zigbee2mqtt/external_converters/encrypted_gp_switch.js` in home assistant supervised. Else its `data/external_converters/encrypted_gp_switch.js` I assume

    ```javascript
    // source https://gist.github.com/dduransseau/6958ec53f6e8665766903e69146285d3
    const exposes = require('zigbee-herdsman-converters/lib/exposes');
    const utils = require('zigbee-herdsman-converters/lib/utils');
    const e = exposes.presets;
    
    
    const fzLocal  = {
        cluster: "greenPower",
        type: ["commandNotification", "commandCommissioningNotification"],
        convert: (model, msg, publish, options, meta) => {
            const commandID = msg.data.commandID;
            if (utils.hasAlreadyProcessedMessage(msg, model, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) return;
            if (commandID === 224) return; // Skip commissioning command.
    
            const lookup = {
                32: "press_1",
                33: "press_2",
                17: "press_3",
            };
    
            const action = lookup[commandID] !== undefined ? lookup[commandID] : `unknown_${commandID}`;
            return {action};
        },
    };
    
    const definition = {
        zigbeeModel: ['GreenPower_2'],
        model: 'GreenPower_2',
        vendor: '',
        description: 'Automatically generated definition',
        extend: [],
        meta: {},
        fromZigbee: [fzLocal],
        exposes: [
            e.action([
                "press_1",
                "press_2",
                "press_3",
            ]),
        ],
    };
    
    module.exports = definition;
    ```

2. Delete device if paired already

3. Restart Zigbee2mqtt addon or service

4. Pair the device by holding pair button and pressing one the switch buttons. (Make sure to match Zigbee channel on on device, 0 = channel 11 by default)

5. After interview, these events should show up then I press the buttons (I have 2 gang one):

```txt
info 2025-04-22 08:29:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c8a27f72', payload '{"action":"press_1","linkquality":213}'
info 2025-04-22 08:29:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c8a27f72/action', payload 'press_1'
info 2025-04-22 08:29:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c8a27f72', payload '{"action":"press_2","linkquality":215}'
info 2025-04-22 08:29:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c8a27f72/action', payload 'press_2'
````

closed 2025-04-22 by choovick

ofershem, 2025-05-09:
Thank you for this, it helped me a lot.
Can you please mention the zigbee coordinator you are using? Is it Sonoff ZBDongle-E by any chance?
I am using Sonoff ZBDongle-p, but I only got it to work in 'permit join' mode using a different router I have in my network.

choovick, 2025-05-09:
> [@choovick](https://github.com/choovick) Thank you for this, it helped me a lot. Can you please mention the zigbee coordinator you are using? Is it Sonoff ZBDongle-E by any chance? I am using Sonoff ZBDongle-p, but I only got it to work in 'permit join' mode using a different router I have in my network.

Casa_SkyConnect, here is my config for it:

```
serial:
  port: >-
    /dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_ce8aafedd760ed11a7b4cae05720eef3-if00-port0
  adapter: ember
  rtscts: true
```

tiagores, 2025-06-09:
Thanks for these instructions @choovick 
It was actually the first time I managed to get the device out of the "unsupported" state. However, I am also having the same issue as @socram70 and the buttons only work when the device is in pairing mode. Did someone find a solution for this?

socram70, 2025-07-27:
There's a solution, but unfortunately not with the SLZB-06. You can use SLZB-06M for this and according to the blog post and discussion here, this will work for you: https://smarthomescene.com/reviews/moes-self-powered-wireless-zigbee-switch-review/#comment-75646

choovick, 2025-07-27:
> There's a solution, but unfortunately not with the SLZB-06. You can use SLZB-06M for this and according to the blog post and discussion here, this will work for you: https://smarthomescene.com/reviews/moes-self-powered-wireless-zigbee-switch-review/#comment-75646

[image]

 🤞

socram70, 2025-07-28:
Yes, I saw that too, thanks, but since the device is out for a while now, I don't expect this to happen any time soon. I also have some Hue bulbs in use, but this also doesn't help as translator.

nh2, 2026-10-02:
> It only works as long as zigbee2mqtt is in pairing mode. It's then recognizing the buttons as it should. As soon as it's out of pairing mode, it doesn't work anymore. Also, in mqtt, the buttons are also not recognized. Is it only me with this issue?

I looked into this as well and have the same problem with the Sonoff ZBDongle-P as coordinator: Only pairing mode shows the button presses.

That is the case even after adding IKEA Tradfri [LED2103G5](https://www.zigbee2mqtt.io/devices/LED2103G5.html) bulb as the device to pair to, to forward ZGP (Zigbee Green Power) messages. The Gemini 3.8 Flash LLM claims this is because the the bulbs can only forward the signal as Groupcast and the ZBDongle-P discards such Groupcast messages unless it's in pairing mode.

So while the comment on https://smarthomescene.com/reviews/moes-self-powered-wireless-zigbee-switch-review/#comment-67214 says

> Sonoff ZBDongle-E is EFR32MG21. (Silicon Labs)
> Sonoff ZBDongle-P is CC2652P. (Texas Instruments)
> You should have no trouble with the first, but the second one does not fully support ZGP (yet). At least, not without a translator like Hue or IKEA.

I find that it doesn't work with ZBDongle-P even with the mentioned translator device.

---

Would be happy to be proven wrong though.