Koenkk/zigbee2mqtt #27184: Xiaomi LYWSD03MMC-z - spikes after upgrade to 3001-0126

dgrazvan, 2025-04-22:
### What happened?

After upgrade over OTA from 3001-0123 to 3001-0126 spikes appeared as in the image (blue). Yellow is a sensor located in the same area (<20cm). The spikes appear randomly. I haven't found a repeating pattern.

[image]

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

Add-on on Home Assistant - virtual machine on proxmox

mrmaximas, 2025-04-22:
https://github.com/pvvx/ZigbeeTLc/issues

dgrazvan, 2025-04-22:
> https://github.com/pvvx/ZigbeeTLc/issues

Thx. I open https://github.com/pvvx/ZigbeeTLc/issues/193

closed 2025-06-29 by github-actions[bot]