Koenkk/zigbee2mqtt #27185: battery property is null for all my device

ghedolo, 2025-04-22:
### What happened?

after upgrade to 2.2.1-dev  all my device have a lot of properties null : battery is the worst case

follows some example of waht i found on mqtt   (the data are grabbed with mqtt -explorer 0.40.-beta6)

example ikea e2134 Fw:1.0.64
{
  "battery": null,
  "identify": null,
  "illuminance": 111,
  "last_seen": "2025-04-22T13:47:10+02:00",
  "linkquality": 47,
  "occupancy": null,
  "update": {
    "installed_version": -1,
    "latest_version": -1,
    "state": null
  }
}

example sonooff SNZB-02D fw 2.3.0

{
  "battery": null,
  "comfort_humidity_max": null,
  "comfort_humidity_min": null,
  "comfort_temperature_max": null,
  "comfort_temperature_min": null,
  "humidity": 34.6,
  "humidity_calibration": null,
  "last_seen": "2025-04-22T13:47:59+02:00",
  "linkquality": 51,
  "temperature": null,
  "temperature_calibration": null,
  "temperature_units": null,
  "update": {
    "installed_version": -1,
    "latest_version": -1,
    "state": null
  }
}

### What did you expect to happen?

have a value in the battery property

### How to reproduce it (minimal and precise)

just start z2m : the battery are always null 

### Zigbee2MQTT version

2.2.1-dev commit: faa6ad9c

### Adapter firmware version

20240710

### Adapter

sonoff zigbee 3.0

### Setup

rasparry pi / debian 12.9

### Debug log

[log.log](https://github.com/user-attachments/files/19849120/log.log)

chefhb88, 2025-04-24:
Hi,
i think same problem as mine. -> [https://github.com/Koenkk/zigbee2mqtt/issues/27020]

ghedolo, 2025-05-11:
an update .
Upgrade Zigbee2MQTT version to 
[2.3.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.3.0) commit: [9d1427c1](https://github.com/Koenkk/zigbee2mqtt/commit/9d1427c1)
Coordinator type ZStack3x0
Coordinator revision 20240710
Frontend version 0.9.10
zigbee-herdsman-converters version 23.36.0
zigbee-herdsman version 4.0.0

but the battery level is sitll null on all device
An exaple: ikea [E2134]
{
  "battery": null,
  "identify": null,
  "illuminance": null,
  "last_seen": "2025-05-11T16:44:34+02:00",
  "linkquality": 72,
  "occupancy": false,
  "update": {
    "installed_version": -1,
    "latest_version": -1,
    "state": null
  }
}

bests

luca

ghedolo, 2025-06-08:
today i update to [2.4.0] commit: [bdb94da4],
but the battery of all my device a still "null". 
i tried to pair again some of them, but the battery still "null"