Koenkk/zigbee2mqtt #27186: ikea E2001/E2002 does not work animore

ghedolo, 2025-04-22:
### What happened?

every time click a key a got: z2m: Invalid message 'undefined', skipping... on the frontend

### What did you expect to happen?

got somting on  the mqtt

### How to reproduce it (minimal and precise)

click one key on the device

### Zigbee2MQTT version

2.2.1-dev commit: faa6ad9c

### Adapter firmware version

20240710

### Adapter

sonoff dongle 3.0

### Setup

rasperry and debian 12.9

### Debug log

[log.log](https://github.com/user-attachments/files/19849313/log.log)

ghedolo, 2025-04-24:
I use only node-red, no automation at all.
the malformed msg apper every time a key is pressed on ikea button

chris-1243, 2025-04-24:
> [2025-04-22 13:58:59] debug: 	z2m:mqtt: Received MQTT message on 'z2m/LC*/set' with data '{"state":Off}'
[2025-04-22 13:58:59] error: 	z2m: Invalid message 'undefined', skipping...

From your log, this payload is wrong

It should be
```
{"state": "OFF"}
```

ghedolo, 2025-04-25:
oops.. it's my fault. in a library update I didn't notice that a ".toUpperCase()" disappeared.
I had been working  about the problem of the disappearance of the "action" by changing many things before realizing that it was a radical change in the implementation. (but joyfull in the dev brench the action are working well)
thanks a lot

closed 2025-04-25 by ghedolo