Koenkk/zigbee2mqtt #27187: Update from v 2.1.1: failed, zigbee2mqtt.service: Start request repeated too quickly

Nick4-1, 2025-04-22:
### What happened?

Since running 2.1.1 for some time, I cannot update anymore

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

### Setup

Z2M on Proxmox (LXC) Bookworm, up-to-date
Originally Installed with tteck's script quite some time ago
2vCPU, 1024Mb RAM

### Debug log

`systemctl status zigbee2mqtt.service
x zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-04-21 16:47:43 CEST; 3min 48s ago
   Duration: 1.430s
    Process: 252 ExecStart=/usr/bin/npm start (code=exited, status=1/FAILURE)
   Main PID: 252 (code=exited, status=1/FAILURE)
        CPU: 1.703s

Apr 21 16:47:43 z2m systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Apr 21 16:47:43 z2m systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Apr 21 16:47:43 z2m systemd[1]: zigbee2mqtt.service: Consumed 1.703s CPU time.
Apr 21 16:47:43 z2m systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Apr 21 16:47:43 z2m systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 21 16:47:43 z2m systemd[1]: Failed to start zigbee2mqtt.service - zigbee2mqtt.

sudo journalctl -u zigbee2mqtt.service -f
Apr 21 17:05:48 z2m npm[265]:     at ModuleLoader.import (node:internal/modules/esm/loader:335:23)
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Consumed 1.689s CPU time.
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Apr 21 17:05:48 z2m systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Consumed 1.689s CPU time.
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Apr 21 17:05:48 z2m systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 21 17:05:48 z2m systemd[1]: Failed to start zigbee2mqtt.service - zigbee2mqtt.`

mrmaximas, 2025-04-23:
The first thing I would recommend is to flash newest firmware (i'm using stable 20240710) here
https://dongle.sonoff.tech/sonoff-dongle-quick-flasher/

Nick4-1, 2025-04-27:
Hi, thank you for jumping in!

TBH: I have been holding off to update the FW of my coordinator out of fear to break to much and ending up with a house where nothing works anymore.

closed 2025-07-05 by github-actions[bot]