Koenkk/zigbee2mqtt #27201: Aqara US H2 Two Rocker (WS-K02E) No Bottom Action

Anon33M00se, 2025-04-23:
### What happened?

I bought a couple of US/American Aqara H2 units with neutral and thread capable, but they are flashed to Zigbee mode. 

They were not supported (not recognized) in current main docker images of zigbee2mqtt, so I changed over to running the latest-dev version of the image. The switch now appears and gets configured by zigbee2mqtt. But there is only are only actions for the top powered button ("Top turned on", "Top turned on or off", etc.) and no actions for the bottom button. And if I Listen to all events in developer tools, nothing happens when I mess with the bottom button.

Happy to manually edit the lumi.ts file inside my running docker version to mess with things if that would prove helpful for debugging. 

### What did you expect to happen?

Bottom button would generate some kind of event. 

### How to reproduce it (minimal and precise)

Pair Aqara WS-K02E, push bottom button/switch.

### Zigbee2MQTT version

2.2.1-dev commit: faa6ad9ce883644eada82c04a9c11d41e62b4799

### Adapter firmware version

    8.0.2 [GA]  EmberZNet

### Adapter

SLZB-06M

### Setup

docker image koenkk/zigbee2mqtt

davidlday, 2025-04-28:
The original PR had an incorrect number of endpoints (sloppy work on my part). It was corrected in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9150. I suspect the second PR hasn't made it into the dev image yet, but I'll see what I can find out.

davidlday, 2025-04-28:
@Anon33M00se - okay, v23.32.0 of zigbee-hersman-converters appears to have the correct number of endpoints in it. I'll have to take a look at my setup and see what else might be causing it. In the meantime, can you make sure the firmware is up to date? Current firmware version my my switch is `20241209`.

Anon33M00se, 2025-04-29:
I am also running the 20241209 firmware. Here is more detail (from the Zigbee2MQTT web interface):

| Item                      |                 Value                       |
|-------------------------|-----------------------------------------|
| Friendly name    |   Dual Switch (B0rked)          |
| Description         |                                                  |
| Last seen             |                        N/A                   |
| Availability           |       Disabled                           |
|Device type                          |        Router |
|Zigbee Model                     |         lumi.switch.agl004 |
|Zigbee Manufacturer     |            Aqara  |
|Description |Light Switch H2 US (double rocker)  |
| Support status | Supported |
|IEEE Address | 0x54ef4410011b121a |
| Network address | 0xD0C4 / 53444 |
| Firmware build date | 20241209|
|Manufacturer |[Aqara](https://www.zigbee2mqtt.io/supported-devices/#v=Aqara) |
|Model | [WS-K02E](https://www.zigbee2mqtt.io/devices/WS-K02E.html#aqara-ws-k02e) |
| Power | |
| Interview completed |True |

aarosso, 2025-05-06:
I have exactly the same issue. Same firmware. I have not managed to get the bottom button to do anything.

I understand that 'Top' is the top button and 'Wireless' is the bottom button (the one that is not connected to the relay). 

Maybe un-related but I do see the some mention of a "multi click mode for wireless button" in the exposes tab but this fails when I try to enable it fails with:

[2025-05-06 14:31:12] error: 	z2m: Publish 'set' 'multi_click' to '0x54ef441....' failed: 'Error: ZCL command 0x54ef441.../2 manuSpecificLumi.write({"646":{"value":1,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4447,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

I also see an entity for 'multi click wireless' but that doesn't seem to do anything. 

I suspect that the wireless button has to be enabled somehow for it to send events.

davidlday, 2025-05-08:
@Anon33M00se @aarosso - apologies for the delay. I've updated my Zigbee2MQTT HA Add-on to the latest version and removed the external converter I've been using. Here's a copy of the external converter I used previously (as a txt file) for reference.

[lumi-switch-agl004.js.txt](https://github.com/user-attachments/files/20095330/lumi-switch-agl004.js.txt)

My switch works the same as it did with the external converter. The only difference I see is in the Coordinator type. I'm using ZStack3x0 vs @Anon33M00se EmberZNet. I have no idea if that would make a difference.

I seem to recall jumping through some hoops when switching switching the firmware from Thread to Zigbee, but I can't recall for sure. When I get some more time, I'll do a factory reset on the switch and start from scratch.

@aarosso - The multi click didn't work for me either. Noted that in the original [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9139). I plan to look into it, but not sure where to begin debugging that one since that's from the underlying libraries.

Logs showing what I get on the topic
```
info 2025-05-07 21:43:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/office/light_switch/action', payload 'single_top'
info 2025-05-07 21:43:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/office/light_switch/action', payload 'single'
```

Note for both switches, I only get single, but I do get it for both.

For reference, my setup:

Home Assistant
* Core: 2025.5.0
* Supervisor: 2025.04.1
* Operating System: 15.2
* Frontend: 20250507.0 

Zigbee2MQTT version: [2.3.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.3.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type: ZStack3x0

Coordinator revision: 20240710

Frontend version: 0.9.10

zigbee-herdsman-converters version: 23.36.0

zigbee-herdsman version: 4.0.0

aarosso, 2025-05-08:
@davidlday - I’m not sure what i did since I’ve been tinkering at this for the last few days but now it seems to work! I see what you’re seeing…. I see a `single_top` event for the top button and a `single` even for the bottom button. I suspect what helped was the upgrade i just did of core from 2025.4.4 to 2025.5.0 but i don’t know for sure.

Just for reference, i had to install the Aqara app and create an account to then be able to connect to the device and switch it to zigbee mode. From there it seems like you need an Aqara hub to do anything else.

davidlday, 2025-05-09:
@aarosso - I'll confirm I also had to install the Aqara app to switch it to zigbee, but no other options to configure it. I may have to pick up an Aqara hub to troubleshoot further on the multi click option.

modem7, 2025-05-10:
The issue also seems to be present in the [Aqara WS-K07E](https://www.zigbee2mqtt.io/devices/WS-K07E.html#aqara-ws-k07e) version (EU) - would it be worthwhile raising a new ticket for this? I don't want to hijack or derail.

aquadan88, 2025-05-20:
I’m also experiencing this issue as well. Has there been any update to this?

aquadan88, 2025-07-20:
Anyone gotten this to work? I reverted back to Matter for the time being until this is resolved.

modem7, 2025-07-20:
Can we remove the stale label please? This is not resolved.

Holbert, 2025-08-07:
Hi, also trying to get this bottom button working. I just setup Z2M to use this device so don't have a ton of experience with Z2M. Happy to help get this working by providing logs or making changes where possible. 

Running latest version of Zigbee2MQTT (2.6.0) and latest Home Assistant (2025.8.0).
Using SLZB-MR1 - Radio 1 EFR32MG21

Holbert, 2025-08-08:
Log for when setting setting multiclick to true:

```logs
[2025-08-08 17:20:53] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/EmmySwitch/set' with data '{"multi_click_wireless":true}'
[2025-08-08 17:20:53] debug:    z2m: Publishing 'set' 'multi_click' to 'EmmySwitch'
[2025-08-08 17:20:53] debug:    zh:controller:endpoint: ZCL command 0x54ef4410012b153a/2 manuSpecificLumi.write({"646":{"value":2,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4447,"writeUndiv":false})
[2025-08-08 17:20:53] debug:    zh:ember: ~~~> [ZCL to=0x54ef4410012b153a:16579 apsFrame={"profileId":260,"clusterId":64704,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":true},"manufacturerCode":4447,"transactionSequenceNumber":8,"commandIdentifier":2}]
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=69 Len=31]
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=5 frmRx=5](ackRx=5)
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=5 frmTx=6)
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=5](frmRx=5) Added to rxQueue
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=6)
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=69 Len=10]
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=250 messageTag=35 status=OK]
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=6](frmRx=6) Added to rxQueue
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=6)
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=69 Len=18]
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: ezspIncomingRouteRecordHandler: source=16579 sourceEui=0x54ef4410012b153a lastHopLqi=136 lastHopRssi=-77 relayCount=0 relayList=
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=7](frmRx=7) Added to rxQueue
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=6)
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=69 Len=45]
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: ezspIncomingMessageHandler: type=UNICAST apsFrame={"profileId":260,"clusterId":64704,"sourceEndpoint":2,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":119} packetInfo:{"senderShortId":16579,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":8,"lastHopLqi":136,"lastHopRssi":-77,"lastHopTimestamp":238312701} messageContents=1c5f110804868602
[2025-08-08 17:20:53] debug:    zh:controller: Received payload: clusterID=64704, address=16579, groupID=0, endpoint=2, destinationEndpoint=1, wasBroadcast=false, linkQuality=136, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4447,"transactionSequenceNumber":8,"commandIdentifier":4},"payload":[{"status":134,"attrId":646}],"command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}}
[2025-08-08 17:20:53] debug:    zh:controller:endpoint: Error: ZCL command 0x54ef4410012b153a/2 manuSpecificLumi.write({"646":{"value":2,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4447,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:372:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1005:26)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at runNextTicks (node:internal/process/task_queues:69:3)
    at processImmediate (node:internal/timers:453:9)
    at Endpoint.write (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:417:9)
    at Object.toZigbee.convertSet (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2539:27)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17)
[2025-08-08 17:20:53] error:    z2m: Publish 'set' 'multi_click' to 'EmmySwitch' failed: 'Error: ZCL command 0x54ef4410012b153a/2 manuSpecificLumi.write({"646":{"value":2,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false [... cut ...]
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=0](frmRx=0) Added to rxQueue
[2025-08-08 17:20:53] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=6)
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=69 Len=26]
[2025-08-08 17:20:53] debug:    zh:ember:ezsp: ezspMessageSentHandler: status=OK type=DIRECT indexOrDestination=16579 apsFrame={"profileId":260,"clusterId":64704,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":250} messageTag=35

Holbert, 2025-08-09:
I'm just getting started with Zigbee2MQTT so may have not done this correctly, but I was able to get this working by setting up an external converter( thanks @davidlday for posting your  example above). 

I changed the wireless endpoint from 2 to 4 and it is working as expected. Able to enable multiclick and then see actions for single_wireless, double_wireless, and hold and release.

`m.deviceEndpoints({endpoints: {top: 1, wireless: 4}}),`

Koenkk, 2025-08-10:
@Holbert Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/lumi.ts)?

Holbert, 2025-08-11:
Submitted. First time, but I believe it should be correct.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9780