Koenkk/zigbee2mqtt #27204: Fails to start after update from 1.42.0 to 2.2.1 with udev rule naming

TjardE, 2025-04-23:
### What happened?

Fails to start: Docker udev rules gives invalid config in docker

### What did you expect to happen?

normal startup as was before I updated from 1.42.0 to 2.2.1

### How to reproduce it (minimal and precise)

make a udev rule for the USB dongle
SUBSYSTEM=="tty", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", SYMLINK+="ttyUSB-ZIGBEE"

now the device to use in docker-compose.yml is

  devices:
  - "/dev/ttyUSB-ZIGBEE:/dev/ttyUSB-ZIGBEE"

so the device to use in zigbee2mqtt config is 
/dev/ttyUSB-ZIGBEE

This does not seem to work anymore after update, so I had to revert to previous version 1.42.0 where it works.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

 20221226

### Adapter

 zStack3x0 compatible

### Setup

odroid with docker

lnlyssg, 2025-04-24:
Did you check and follow the breaking changes guidance at https://github.com/Koenkk/zigbee2mqtt/discussions/24198?

TjardE, 2025-04-28:
Somehow the adapter type was set to 'auto' in the configuration.yml.
It should be one of the following: zstack, ember, deconz, zigate or zboss.

This part fixed for me. Still not everything is working but this issue can be closed

TjardE, 2025-05-24:
Now that I updated to 2.3.0 everything works as expected.