Koenkk/zigbee2mqtt #27205: Incorrect formaldehyde value interpretation for TS0601_smoke (Tuya) in Zigbee2MQTT

papa-aus-rastatt, 2025-04-24:
### What happened?

**Device model**: TS0601_smoke (Tuya)
**Coordinator**: SMLIGHT SLZB-06P7

**Bug description**:
The `formaldehyd` value from TS0601_smoke is incorrectly shown as a full number (e.g. `4 mg/m³`), although the raw value seems to be in 1/1000 mg/m³.

**Example**:
- Raw value from payload: `60`
- Interpreted value in Zigbee2MQTT: `4` (wrong!)
- Correct value should be: `0.06 mg/m³`

Same issue possibly affects `voc` and `co2` fields too.

**Expected behavior**:
The integration should divide the raw payload by 1000 to show realistic values (like 0.06 instead of 4).

**Steps to reproduce**:
- Pair the device
- Read `formaldehyd` in MQTT topic (e.g. `zigbee2mqtt/hausflur`)
- Compare raw and interpreted value

Thanks for your great work!

### What did you expect to happen?

Update the Sensor-Data Calculation.

### How to reproduce it (minimal and precise)

default Integration.

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

V1.0.6

### Adapter

TS0601_smoke_6.md

### Setup

Add-On in HA / Zigbee2MQTT

mayraost, 2025-06-24:
I have the Same Problem, values are way to high, i should be dead. Found this issue still unresolved, with the Same Problem: https://github.com/Koenkk/zigbee2mqtt/issues/21666

krzkraw, 2025-07-20:
This should be fixed now with [commit in converters](https://github.com/Koenkk/zigbee-herdsman-converters/commit/e31057010670dd6a8cf14b78e1b8a11ea826b9c0)

closed 2025-07-21 by papa-aus-rastatt