Koenkk/zigbee2mqtt #27209: [New device support]: Hue Play Gradient PC LCX007 appears as NOT supported (3x 24"/27")

RemyyB, 2025-04-24:
### Link

https://www.philips-hue.com/de-de/p/lightstrips-pc-play-gradient-lightstrip-3x-24-27/8719514434592

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0x001788010b89e3f9","nwkAddr":29283,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCX007","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20220331","swBuildId":"1.97.3","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1745519362493}

### Zigbee2MQTT version

 2.2.1 commit: a14feef8c13f5ce49b0ac358d98d3234d21d7ff4

### Comments

Hey Koen!

I just received my new Hue Play Gradient Lightstrip for PC (24/27 inch), but it is not supported in Zigbee2MQTT, however I saw that is does have a firmware update, but a bit scared to update due do this not being in the supported state (don't know if it matters?).

Basically a dupe of: [#20453](https://github.com/Koenkk/zigbee2mqtt/issues/20453) and [#20146](https://github.com/Koenkk/zigbee2mqtt/issues/20146) but for the longer version (3x 24/27 inch variant). The external definition is not complete I think because it's missing the other fields because it's individually addressable.

Thank you very much! I really appreciate your work on this!
If you need more information please let me know!

Links to zigbee2mqtt website on the same strips if it helps:
1. [929003498601](https://www.zigbee2mqtt.io/devices/929003498601.html)
2. [8719514434479](https://www.zigbee2mqtt.io/devices/8719514434479.html)
3. [8719514434530](https://www.zigbee2mqtt.io/devices/8719514434530.html)

### External definition

```shell
import * as philips from 'zigbee-herdsman-converters/lib/philips';

export default {
    zigbeeModel: ['LCX007'],
    model: 'LCX007',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I can change the brightness, color temperature etc., same as other Philips Hue devices, but not the addressable part and the effects that are missing on this specific strip.

chris-1243, 2025-04-24:
It should be available in `latest-dev` when merged or in the next official release

RemyyB, 2025-04-25:
Awesome, thank you @chris-1243 ! I will try it.

Koenkk, 2025-04-26:
Will be supported in the 1 May release!

closed 2025-04-26 by Koenkk