Koenkk/zigbee2mqtt #27210: Third reality sensor 3RDTS01056Z showing as mains powered in z2m 2.2.1

smuskiew, 2025-04-25:
### What happened?

Third Reality tilt sensor 3RDTS01056Z is showing as mains powered in Zigbee2MQTT 2.2.1 when it should be a battery device.

It _appears_ that this was previously addressed in 1.x in [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/23694), perhaps it didn't make it into 2.x or somehow got undone?

(This is a new device that I added to my 2.2.1 z2m as of the other day so I can't personally speak as to whether the original issue was really resolved in 1.x or not, just found it when looking thru prior issues for this device model number)

### What did you expect to happen?

Expected it to show as a battery powered device

### How to reproduce it (minimal and precise)

Add the 3RDTS01056Z device to a Zigbee2MQTT 2.2.1 install.  Observe the UI power.

(Note: I did have an issue joining the device the first time around, had to reattempt.  Unsure if this contributed to the problem or not)

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240710

### Adapter

Sonoff ZBDONGLE-P

### Setup

Plain on docker container

chris-1243, 2025-04-25:
It should be fixed. You will need to wait until the fix is reviewed and approved by **Koenkk**.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9246

smuskiew, 2025-04-26:
@chris-1243 Awesome thanks for the quick response!!

rbeuque74, 2025-05-12:
Hello,
Was looking for the same problem, seems that it has been addressed with release 2.3.0 if you want to close it.

Romain

closed 2025-05-13 by Koenkk

smuskiew, 2025-05-26:
Reconfigure gave me this error in z2m:

```
zigbee2mqtt  | [2025-05-26 14:58:35] error: 	z2m: Failed to configure 'Basement/Garage/Doors/left', attempt 4 (Error: Bind 0x282c02bfffeee4a3/1 genPowerCfg from '0x00124b0024c26205/1' failed (AREQ - ZDO - bindRsp after 10000ms)
zigbee2mqtt  |     at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
zigbee2mqtt  |     at func (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:419:47)
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:105:5)
zigbee2mqtt  |     at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
zigbee2mqtt  |     at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:424:43)
zigbee2mqtt  |     at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:301:16)
zigbee2mqtt  |     at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:580:30)
zigbee2mqtt  |     at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:13)
zigbee2mqtt  |     at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:209:17)
zigbee2mqtt  |     at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21))
```

I did a restart z2m anyway but even after that it is still showing as mains powered.

(The device is working otherwise, it still reports its status correctly when the garage door is opened or closed.)

I did notice that the first time it sent a status after Z2M was restarted that it was also complaining about `UNSUPPORTED_ATTRIBUTE`, here is the log entry for that:

```
zigbee2mqtt  | [2025-05-26 15:04:07] error: 	z2m: Failed to configure 'Basement/Garage/Doors/left', attempt 1 (Error: ZCL command 0x282c02bfffeee4a3/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt  |     at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:363:28)
zigbee2mqtt  |     at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:987:26)
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:105:5)
zigbee2mqtt  |     at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:707:9)
zigbee2mqtt  |     at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:17)
zigbee2mqtt  |     at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:209:17)
zigbee2mqtt  |     at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
zigbee2mqtt  |     at Configure.configure (/app/lib/extension/configure.ts:124:13)
zigbee2mqtt  |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

rbeuque74, 2025-05-26:
I also spotted quickly this log message, during multiple device deletion and rejoining:

```
pi@raspberrypi:~ $ journalctl  -ru zigbee2mqtt.service --since=yesterday | grep 0xffffb40e0601af54 | grep "ZigBee specifica"
mai 26 21:02:43 raspberrypi node[16337]: [2025-05-26 21:02:43] info:         zh:controller:device: Device '0xffffb40e0601af54' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
```

This only happened once, even if I deleted and rejoin the device multiple times, not sure it is actually relevant to this issue, but might help.

Romain

chris-1243, 2025-05-26:
It seems your device does not accept "batteryPercentageRemaining" as an attribute while configuring the reporting. The fix I provided does not change this.

Does anyone of you could add this code as an external converter with `*.mjs` extension

```javascript
import {Zcl} from "zigbee-herdsman";
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
    zigbeeModel: ["3RDTS01056Z"],
    model: "3RDTS01056Z",
    vendor: "Third Reality",
    description: "Garage door tilt sensor (test)",
    extend: [
        m.battery({percentageReporting: false}),
        m.forcePowerSource({powerSource: "Battery"}),
        m.iasZoneAlarm({zoneType: "contact", zoneAttributes: ["alarm_1", "battery_low"]}),
        m.deviceAddCustomCluster("3rGarageDoorSpecialCluster", {
            ID: 0xff01,
            manufacturerCode: 0x1407,
            attributes: {
                delayOpenAttrId: {ID: 0x0000, type: Zcl.DataType.UINT16},
                zclCabrationAttrId: {ID: 0x0003, type: Zcl.DataType.UINT16},
            },
            commands: {},
            commandsResponse: {},
        }),
    ],
    ota: true,
};
```

rbeuque74, 2025-05-26:
Thanks !
I tested it, and I no longer have the stack traces when interacting with the device.
Also, device is well reported as battery now !

chris-1243, 2025-05-26:
Do you have a value for the battery in the expose tab?