Koenkk/zigbee2mqtt #27212: ZY-M100-S_2 (_TZE284_iadro9bf) no "Presence" and no "Target distance"

Softas2024, 2025-04-25:
### What happened?

The ZY-M100-S_2 (_TZE284_iadro9bf) does not recognize the "Presence" and "Target distance"

    "detection_delay": 5.1,
    "fading_time": 32.5,
    "illuminance": 282,
    "last_seen": "2025-04-25T13:53:34+02:00",
    "linkquality": 140,
    "maximum_range": 3.45,
    "minimum_range": 0.75,
    "radar_sensitivity": 5,
    "presence": null,
    "target_distance": null

### What did you expect to happen?

Normal, useful functionality.

### How to reproduce it (minimal and precise)

immer Presence: null
immer Target distance: null

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Addon on Homeassistant on Beelink MINI-S12 Pro

### Debug log

info 2025-04-25 14:04:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZY-M100-S_2', payload '{"detection_delay":5.1,"fading_time":32.5,"illuminance":248,"last_seen":"2025-04-25T14:04:06+02:00","linkquality":112,"maximum_range":3.45,"minimum_range":0.75,"presence":null,"radar_sensitivity":5,"target_distance":null}'

wolfwander, 2025-04-26:
I have an issue with the same type of sensor where the illuminance state is not updated. I am keeping my Zigbee2MQTT in Docker image 2.2.0 since this issue was introduced in 2.2.1.

Softas2024, 2025-04-27:
No, not yet. I'm being a bit cautious because everything else is working fine. I tried pairing it under ZHA (yes, I have both running), and there is a provisional quirk there where presence is detected, but illuminance is not. Maybe a proper solution will come soon.

lukaszzyla, 2025-04-28:
same issue as described in first post.
Presence detection doesn't work.
I have just bought this sensor so never had it working. I suspect there might be one more reading of the sensor missing though.
In many other posts I see this sensor provides motion sensor together with presence. In my case it is only presence and it does not work.
I was able to make Target distance working by setting its values in settings.
```{
    "detection_delay": 0.2,
    "fading_time": 0.5,
    "illuminance": 15,
    "linkquality": 132,
    "maximum_range": 6.45,
    "minimum_range": 0,
    "radar_sensitivity": 7,
    "target_distance": 5.5,
    "presence": null
}```

lukaszzyla, 2025-04-29:
@wolfwander I took the extra step and installed 2.2.0 zigbee2mqtt in docker.
on 2.2.0 it is even more funny!
```
{
    "detection_delay": 0.1,
    "fading_time": 35,
    "illuminance": 85,
    "linkquality": 140,
    "maximum_range": 6,
    "minimum_range": 0.6,
    "presence": true,
    "radar_sensitivity": 7,
    "target_distance": 5.37
}
```

Presence says true, but it is just true all the time ;-)

[image]

It doesn't detect any motion, there is no sensor that would look like one decting motion.
Presence is just frozen at "Home" no matter anything moves there or not - Presence state is always true.

Softas2024, 2025-05-01:
Updated to version 2.3.0-1 today. Unfortunately, illuminance is still the only function that works.

lukaszzyla, 2025-05-06:
Illuminance works for me too, but reports small values.
Target disctance works and values change but they do not seem to be real/correct.
What I mean that it reports if I move it closer to the wall but shows unreal distance.
But for me the presence was the only function I bought it for. I didn't have any motion sensor on stairway and wanted to swith on stairway leds with it.
my problem here:
I am in the unlucky team then, because for me it is still the same.
The funny thing is that illuminance was always working for me together with distance.
but the presence is either true or none
I have removed and readded the device so many times an many different versions of z2m.
I was hoping that 2.3.0 fixes it for me like for rest of you but unfortunatelly - no difference.
mqtt broker:

[image]

z2m:
When I add the device it is being interviewed and reports success.
But then presence is frozen on null or true.
nobody's at home for an hour or so and:

[image]

[image]

[image]

[image]

Illuminance levels are far away from my knx sensors that report several thousands lux compared to 2-3 lx from tuya.
Any ideas what to do next?
It never worked for me as I bought it 2 or 3 weeks ago.
Do you think this could be faulty device?
I do not have tuya gateway to check with native app.

klausmcm, 2025-05-10:
Same issue as OP (I'm on **2.3.0**)

```
[2025-05-09 16:55:44] info: 	z2m: Device '0xa4c138d5a42447ff' is supported, identified as: Tuya Mini human breathe sensor (ZY-M100-S_2)
[2025-05-09 16:55:44] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Mini human breathe sensor","exposes":[{"access":1,"description":"Measured illuminance","label":"Illuminance","name":"illuminance","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Indicates whether the device detected presence","label":"Presence","name":"presence","property":"presence","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Distance to target","label":"Target distance","name":"target_distance","property":"target_distance","type":"numeric","unit":"m"},{"access":3,"description":"sensitivity of the radar","label":"Radar sensitivity","name":"radar_sensitivity","property":"radar_sensitivity","type":"numeric","value_max":9,"value_min":0,"value_step":1},{"access":3,"description":"Minimum range","label":"Minimum range","name":"minimum_range","property":"minimum_range","type":"numeric","unit":"m","value_max":9.5,"value_min":0,"value_step":0.15},{"access":3,"description":"Maximum range","label":"Maximum range","name":"maximum_range","property":"maximum_range","type":"numeric","unit":"m","value_max":9.5,"value_min":0,"value_step":0.15},{"access":3,"description":"Detection delay","label":"Detection delay","name":"detection_delay","property":"detection_delay","type":"numeric","unit":"s","value_max":10,"value_min":0,"value_step":0.1},{"access":3,"description":"Fading time","label":"Fading time","name":"fading_time","property":"fading_time","type":"numeric","unit":"s","value_max":1500,"value_min":0.5,"value_step":1},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZY-M100-S_2","options":[{"access":2,"description":"Calibrates the illuminance value (percentual offset), takes into effect on next report of device.","label":"Illuminance calibration","name":"illuminance_calibration","property":"illuminance_calibration","type":"numeric"}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c138d5a42447ff","ieee_address":"0xa4c138d5a42447ff","status":"successful","supported":true},"type":"device_interview"}'
```
---
```
{
    "illuminance": 57,
    "linkquality": 83,
    "maximum_range": 4.8,
    "minimum_range": 4.95,
    "radar_sensitivity": 5,
    "detection_delay": null,
    "fading_time": null,
    "presence": null,
    "target_distance": null
}
```

Previously I was on **2.1.3**. Result of pairing then

```
[2025-04-11 00:07:46] info:     zh:controller: Succesfully interviewed '0xa4c138d5a42447ff'
[2025-04-11 00:07:46] info:     z2m: Successfully interviewed '0xa4c138d5a42447ff', device has successfully been paired
[2025-04-11 00:07:46] warning:  z2m: Device '0xa4c138d5a42447ff' with Zigbee model 'TS0601' and manufacturer name '_TZE284_iadro9bf' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
[2025-04-11 00:07:46] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d5a42447ff","ieee_address":"0xa4c138d5a42447ff","status":"successful","supported":false},"type":"device_interview"}'
```

theccm, 2025-11-01:
I think presence flag is broken since this commit : 

https://github.com/Koenkk/zigbee-herdsman-converters/commit/e0d8975a98f49860b8e8dee85cae171632fbdf51

GiuseppeP96, 2025-11-02:
Target distance is not working for me too. But mine is also periodically becoming unavailable. Why is that?

[image]

zango-me, 2025-11-04:
I got 2 of these sensors and one is ONLY reporting illuminance, the other is ONLY reporting presence (bool), none is reporting distance.

It gets detected as [ZY-M100-S_2](https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html#tuya-zy-m100-s_2)

Zigbee-Modell
TS0601
_TZE284_iadro9bf (Mini human breathe sensor)

Status is 
{
    "linkquality": 140,
    "presence": false
}

and
{
    "illuminance": 10,
    "linkquality": 140
}

(coming from zha, had to patch that for these sensors with a custom config too in the past but this thing just seems broken in z2m...)

klausmcm, 2025-12-29:
**v2.7.1** seems to resolve the issue for me.

Phabbrrysse, 2026-01-16:
Same issue in v2.7.2

closed 2026-10-03 by github-actions[bot]

gd510, 2026-11-05:
> 26\. I'm on version 2.12.1 and this is still not working. It's only showing luminance.

Same here, on version 2.14.3-1.