Koenkk/zigbee2mqtt #27213: [Feature request]: Cover State flip

ThatDeltaGuy, 2025-04-25:
### Is your feature request related to a problem? Please describe

Currently, there is the option for covers to 'invert cover'. This takes in 100% and makes it 0% and vice versa. Unfortunately, the [smartwings blind](https://www.zigbee2mqtt.io/devices/WM25L-Z.html) is also reporting the state inverted and there doesn't seem to be any way to reverse it (It looks like there was a way to reverse it with the remote, but that doesn't work any more). Others have also run into this issue when searching.

### Describe the solution you'd like

Renaming 'inverse cover' to 'inverse position' then adding in an 'inverse state' to switch open and closed around

### Describe alternatives you've considered

There's a Jacky template workaround in home assistant you can use, but it's got its own issues

### Additional context

Could be useful for other blinds in the future too

ThatDeltaGuy, 2026-07-06:
Is there any movement with this?

Great-Chart, 2026-09-02:
You should fine this addressed now
https://github.com/Koenkk/zigbee-herdsman-converters/pull/12707

Others have had success here; refer bottom of https://github.com/Koenkk/zigbee2mqtt/issues/26539