Koenkk/zigbee2mqtt #27220: Aqara contact sensors (MCCGQ11LM) reporting intermittently

RedNo7, 2025-04-26:
### What happened?

Recently my Aqara contact sensors which have been super-reliable for over a year have become very reliable, nit reporting when open and/or not reporting when closed.

A restart of Z2M seems to have them working for 10 mins or so. 
 
- Z2M sees them (they are online)
- Reconfigure button (confirmation prompt) is OK
- no errors in the logs

No cue how to diagnose.  My network has not changed.  Even the contact sensor that is 2m away from my co-ordinator exhibits intermittently reporting being open/closed.

### What did you expect to happen?

I expect Z2M to report when the contact sensor is open/closed.

### How to reproduce it (minimal and precise)

No clue why this is a problem so no clue how to reproduce it.

It can work fine second and literally not the next second.  It is very confusing.  I have too many sensors for this to be a sensor hardware issue (the problem started at the same time).  I have recently updated from an early 2.x release to 2.2.1. Downgrading to 2.1.3 has made no difference.  I will keep rolling backwards to see if there is a more robust build.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240710

### Adapter

Sonoff dongle -P

### Setup

Z2m runs as a docker container on a Pi-4 (DietPi OS).

```
uname -a && cat /etc/issue.ne
Linux homeassistant 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
cat: /etc/issue.ne: No such file or directory
```

### Debug log

will add this later.

zen2, 2025-04-26:
I have 27 of these aqara devices working for years here.  Zigbee2mqtt version: 2.2.1.
But I use a really old firmware version (20221226) on a TI LAUNCHXL-CC1352P-2 adapter because it is stable on my network of 112 devices and I prefer to spent time to read issues related to new firmwares than debug my zigbee network with the whole family on my back ;)

So it seems your problem is related to the firmware version and/or the adapter.

RedNo7, 2025-04-26:
I think the adaptor is solid as everything else works fine (80+ devices).

I think you maybe right about the first ware though as I did update it recently.. Is there a safe way to downgrade?

closed 2025-07-03 by github-actions[bot]